sustainable-computing-io/kepler #29: Resolve Container ID by both PID and Cgroup ID

rootfs, 2022-04-13:
There are two following ways to associate a process to its container ID:
- by reading the process's /proc/pid/cgroup and find the container ID. This requires exposing host's /proc filesystem to Kepler container.
- by using the process's Cgroup ID and resolve it in cgroup FS at /sys/fs/cgroup. This doesn't require host's /proc

The Cgroup ID is the current way in practice. However, CgroupID resolution requires Cgroup V2 support. For environment without Cgroup V2 enabled, a fallthrough to PID resolution is needed.

@marceloamaral

marceloamaral, 2022-04-14:
We can identify if the host has cGroup V2 with `grep cgroup2 /proc/filesystems`. Therefore, the system can identify which version of cGroup the host supports.

However, since we need to customize the Kepler YAML deployment to expose `/proc` to cGroup V1, we can create an operator that configures and deploys Kepler. The operator will introduce a new CRD which will also contain some custom settings.

rootfs, 2022-04-14:
@marceloamaral here is the way from openshift [doc](https://docs.okd.io/latest/post_installation_configuration/machine-configuration-tasks.html#nodes-nodes-cgroups-2_post-install-machine-configuration-tasks)
```
you can verify that cgroups v2 is enabled by checking that the sys/fs/cgroup/cgroup.controllers file is present on the node. This file is created by cgroups v2.
```

rootfs, 2022-04-14:
cc @metacosm, Christophe is working on the operator

marceloamaral, 2022-09-12:
The PR ##152 fixed this.

closed 2022-09-12 by marceloamaral