sustainable-computing-io/kepler #71: Conflict between Kepler-exporter with OVN Kubernetes CNI ovn-master

williamcaban, 2022-06-02:
**Describe the bug**

Kepler cannot run on the OpenShift control plane nodes when OVN Kubernetes is the CNI. OVN Kubernetes `ovn-master` DaemonSet already uses hostPort 9102 as seen in the following extract:

```yaml
apiVersion: apps/v1
kind: DaemonSet
metadata:
  annotations:
    kubernetes.io/description: |
      This daemonset launches the ovn-kubernetes controller (master) networking components.
...
spec:
  selector:
    matchLabels:
      app: ovnkube-master
...
        ports:
        - containerPort: 9102
          hostPort: 9102
          name: https
          protocol: TCP
```

Which is the same port the Kepler exporter daemonset tries to use

```yaml
---
---
apiVersion: apps/v1
kind: DaemonSet
metadata:
  name: kepler-exporter
  namespace: monitoring
spec:
  selector:
    matchLabels:
      app.kubernetes.io/component: exporter
      app.kubernetes.io/name: kepler-exporter
...
        ports:
        - containerPort: 9102
          hostPort: 9102
          name: http
```

So, the kepler-exporter Pods remain in `Pending` state
```bash
kepler-exporter-8gkfs   0/2     Pending   0          15h   <none>          <none>   <none>           <none>
kepler-exporter-8ql8n   0/2     Pending   0          26m   <none>          <none>   <none>           <none>
kepler-exporter-nqxpx   0/2     Pending   0          15h   <none>          <none>   <none>           <none>
```
and the events shows
```bash
7m7s        Warning   FailedScheduling         pod/kepler-exporter-8gkfs   0/6 nodes are available: 1 node(s) didn't have free ports for the requested pod ports, 5 node(s) didn't match Pod's node affinity/selector.
27m         Warning   FailedScheduling         pod/kepler-exporter-8ql8n   0/6 nodes are available: 1 node(s) didn't have free ports for the requested pod ports, 5 node(s) didn't match Pod's node affinity/selector.
27m         Warning   ErrorAddingLogicalPort   pod/kepler-exporter-8ql8n   failed to ensurePod monitoring/kepler-exporter-8ql8n since it is not yet scheduled
7m7s        Warning   FailedScheduling         pod/kepler-exporter-8ql8n   0/6 nodes are available: 1 node(s) didn't have free ports for the requested pod ports, 5 node(s) didn't match Pod's node affinity/selector.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy Kepler to an OpenShift control plane using OVN Kubernetes as the CNI.

**Additional context**
- OpenShift version 4.10.15

Looking at this issue, it might be possible to solve it by having Kepler running privileged (which it needs) but NOT having it running in hostNetwork mode.

rootfs, 2022-06-02:
Ha, port becomes a hot resource :D 
Would you post a fix, it looks a few places need to have the new port:

1) the [arg](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/openshift/openshift-deployment.yaml#L70)
2) the [container port](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/openshift/openshift-deployment.yaml#L76-L77)

(in both openshift and kubernetes deployment)

williamcaban, 2022-06-04:
Addressed with https://github.com/sustainable-computing-io/kepler/commit/62be2a1a6a01499ef0cbe244bbf7ea3e617243c6  by moving the Pod to run without hostNetwork mode. Tested with OpenShift 4.10.15 w/OVN.

closed 2022-06-04 by williamcaban