sustainable-computing-io/kepler #75: Kepler Documentation

husky-parul, 2022-06-06:
**Describe the solution you'd like**
Add documentation on:

- About Kepler
- Describe the modules, bpf, model server, collector, power, attacher etc

husky-parul, 2022-06-06:
We can use one of these as a reference
[Volsync](https://volsync.readthedocs.io/en/stable/)
[Microshift](https://microshift.io/)

husky-parul, 2022-06-06:
@rootfs Please assign the issue to me. For some reason I still can't assign issues to myself.

husky-parul, 2022-06-06:
@rootfs @nikimanoledaki @williamcaban I would appreciate your inputs on which docs would look good for Kepler. I have mentioned two references above. My preference is VolSync. It looks simple, intuitive and easy to navigate.

williamcaban, 2022-06-06:
I guess you mean using the [Sphinx](https://www.sphinx-doc.org/) theme. That's fine with me. I'm happy with anything that can use markdown-type files for quick iteration and easier collaboration on docs.

rootfs, 2022-06-06:
Please use [this repo](https://github.com/sustainable-computing-io/kepler-doc) for documentation. Ideally we have doc to serve the following purposes:
- For developers: an deep dive into the development environment, architecture, methodology, enhancement.
- For end users:  Environment requirement, deployment process, dashboard setup and customization. 
- For integrators: Packaging, API, and extensions.

husky-parul, 2022-06-07:
I am going to follow the this flow

## Contents
### Overview
### Installation
#### Kubernetes
#### OpenShift
###  Usage
#### Developers
#### End Users
#### Integratiors

geertpingen, 2022-06-07:
Hi, checking out this project after Kubecon EU. Im currently using [Scaphandre](https://github.com/hubblo-org/scaphandre), and scanning for alternatives.

Im looking for documentation on what exactly it is you're monitoring, and how you are calculating that, similar to:
https://github.com/hubblo-org/scaphandre/blob/main/docs_src/explanations/how-scaph-computes-per-process-power-consumption.md. Then I noticed this issue, so I would recommend having a section like that in your docs. As a user this is the first thing Im looking for.

husky-parul, 2022-06-07:
Thanks @geertpingen for your suggestions. We are still in early stages but to give some context, we calculate power consumption of running containers in Kubernetes env based on cpu cycles/instruction/time/cache misses etc. Incorporating the details on how it is calculates is still WIP and will be available at our docs page [here](https://sustainable-computing-io.github.io/kepler-doc/)

rootfs, 2022-06-07:
@geertpingen thanks for the pointer, alternative is good for consumers :D 

Kepler and Scaphandre both want to give users an answer on how much energy spent by a workload (process, container, or VM). 

The difference is that, Scaphandre spends a good deal on breaking down the processes' CPU time to attribute the energy to the workloads. Kepler uses CPU time as well as CPU spends time on doing. At the end of the day, not all CPU instruction consumes the same amount of energy.

geertpingen, 2022-06-08:
@husky-parul  @rootfs Thanks for the quick replies! Perhaps this is a question best asked in another setting (are you active on Slack or a similar platform?), but I was wondering how you obtain the coefficients here:
https://github.com/sustainable-computing-io/kepler/blob/a71dce95c1e76d1866f417c1ec8fbf20b58403fe/pkg/model/model.go#L47 (used at https://github.com/sustainable-computing-io/kepler/blob/79055d74185b9fad19dc974e6f65312f453ae72c/pkg/collector/reader.go#L320)

rootfs, 2022-06-08:
@geertpingen slack invite sent. The coefficients are hardcoded at the moment. They will be learned via online training by the model server (still under construction). If you are excited about building a model of your own, [it is a place](https://github.com/sustainable-computing-io/kepler-model-server) for collaboration

closed 2023-05-24 by stale[bot]