sustainable-computing-io/kepler #96: cpu time becomes zero after long run

rootfs, 2022-07-19:
**Describe the bug**
After running for extended hours, the curr_cpu_time becomes zero and never recovers
```json
{
  "__name__": "node_energy_stat",
  "container": "kepler-exporter",
  "cpu_architecture": "Cascade Lake",
  "curr_cache_misses": "10019190",
  "curr_cpu_cycles": "4005839184",
  "curr_cpu_instructions": "5068047632",
  "curr_cpu_time": "0.000000",              <----------- this never recovers
  "curr_energy_in_core": "154835.000000",
  "curr_energy_in_dram": "5.000000",
  "curr_energy_in_gpu": "0.000000",
  "curr_energy_in_other": "0.000000",
  "curr_resident_memory": "10457391104.000000",
  "endpoint": "http",
  "instance": "nvidia-gpu.ibm.cloud",
  "job": "kepler-exporter",
  "namespace": "monitoring",
  "node_name": "xxxx",
  "pod": "kepler-exporter-9k75l",
  "service": "kepler-exporter"
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

rootfs, 2022-07-19:
cpu time is calculated [here](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/reader.go#L284-L290)
cc @marceloamaral

marceloamaral, 2023-05-18:
@rootfs do we still have this problem?

closed 2023-08-28 by stale[bot]