sustainable-computing-io/kepler #120: Pod eCore, eUncore is reported as 0, pod in_core metrics reported as 0

Feelas, 2022-08-31:
**Describe the bug**
In kepler-exporter pod logs, eCore is reported as 0 for all pods.
pod_<curr|total>_energy_in_core_millijoule parameters are correctly sent to Prometheus, but also set to 0 all of the time.

**To Reproduce**
1. Run kepler-exporter
2. Check values of eCore metrics and pod_<curr|total>_energy_in_core_millijoule metrics inside Prometheus.

**Expected behavior**
eCore metrics are not reported as 0, pod_<curr|total>_energy_in_core_millijoule metrics report proper values.

**Additional information**
Kepler-exporter is ran from image sha256:819a0b056f86a754c3b58ef31c3dd2fbcf279dcb02caf7e3bfd8a471683081a6

rootfs, 2022-08-31:
cc @sunya-ch

sunya-ch, 2022-08-31:
Could you check whether the core and uncore energies are reported by RAPL in your file system? 
```bash
cat /sys/class/powercap/intel-rapl/intel-rapl\:0/intel-rapl:*/name
```

note: According to previous PR, instead of implying core from `pkg - dram` or vice versa if dram is not available, I think we should report only if that energy event is available.

Feelas, 2022-08-31:
That's interesting:
`[core@worker2 ~]$ cat /sys/class/powercap/intel-rapl/intel-rapl\:0/intel-rapl:*/name
dram
`

rootfs, 2022-08-31:
@Feelas can you try one more thing:

```sh
cat /sys/class/powercap/intel-rapl*/intel-rapl*/name
```

Here is what i have on my setup
```console
# cat /sys/class/powercap/intel-rapl*/intel-rapl*/name
dram
dram
package-0
package-1
```

sunya-ch, 2022-08-31:
The previous version computes core = package - dram if core event doesn't exist. 
Package energy should be equal to core + dram + uncore.  Uncore energy is omitted.

Feelas, 2022-08-31:
@rootfs got the same output here
```
[core@worker2 ~]$ cat /sys/class/powercap/intel-rapl*/intel-rapl*/name
dram
dram
package-0
package-1
```
Can be confirmed also by inspecting dmesg
```
[   19.513659] intel_rapl_common: Found RAPL domain package
[   19.528089] intel_rapl_common: Found RAPL domain dram
[   19.533705] intel_rapl_common: DRAM domain energy unit 15300pj
[   19.548906] intel_rapl_common: RAPL package-0 domain package locked by BIOS
[   19.566514] intel_rapl_common: Found RAPL domain package
[   19.580837] intel_rapl_common: Found RAPL domain dram
[   19.586482] intel_rapl_common: DRAM domain energy unit 15300pj
[   19.601336] intel_rapl_common: RAPL package-1 domain package locked by BIOS
```

closed 2022-08-31 by rootfs

marceloamaral, 2022-09-09:
@Feelas please see the PR #156

PR #121 (2022-08-31): derive missing energy domain from pkg and other domain
fix #120