sustainable-computing-io/kepler #126: 03-grafana-datasource-define.sh uses deprecated 'oc serviceaccounts get-token'

Feelas, 2022-09-02:
**Describe the bug**
As can be seen in [OpenShift 4.11 release notes](https://docs.openshift.com/container-platform/4.11/release_notes/ocp-4-11-release-notes.html), the dashboard creation command found in `manifests/openshift/dashboard/03-grafana-datasource-define.sh` uses deprecated `oc serviceaccounts get-token grafana-serviceaccount -n monitoring` command. As a result, the command does not return a valid token. This causes the Grafana datasource to not be created.

Example output form the command (using oc cli 4.11):
```
[root@workstation dashboard]$ oc serviceaccounts get-token grafana-serviceaccount -n monitoring
Command "get-token" is deprecated, and will be removed in the future version. Use oc create token instead.
error: could not find a service account token for service account "grafana-serviceaccount"
```

**To Reproduce**
Steps to reproduce the behavior:
1. Use an OpenShift 4.11 cluster
2. Run manifests/openshift/dashboard/03-grafana-datasource-define.sh script and note the output of the script.

**Expected behavior**
Token is created successfully and 03-grafana-datasource-define.sh works properly, creating a Prometheus datasource.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OpenShift 4.11 cluster
- oc tool 4.11.0-rc.1 (any version of 4.11 oc cli can be used)

**Additional context**
I used a straight replacement command `oc create token grafana-serviceaccount --duration 31536000s -n monitoring` to create a 365d token for the dashboard.

rootfs, 2022-09-02:
OpenShift 4.11 is based on kubernetes 1.24, this token issue is similar to #110 
cc @ruomengh

sallyom, 2022-09-02:
will need to update the manifests to create the secret similar to the dev/role.yaml also https://github.com/sustainable-computing-io/kepler/blob/main/dev/role.yaml#L23-#L34

Feelas, 2022-09-27:
Closing due to #128 being merged.

closed 2022-09-27 by Feelas