sustainable-computing-io/kepler #131: panic: inconsistent label cardinality: expected 21 label values but got 20

jichenjc, 2022-09-05:
**Describe the bug**
A clear and concise description of what the bug is.

follow https://github.com/sustainable-computing-io/kepler

after `~/kube-prometheus# kubectl apply -f manifests/`, saw this error

seems something inconsistent ..
```
panic: inconsistent label cardinality: expected 21 label values but got 20 in []string{"system_processes", "system", "containerd", "388246", "2674068", "0", "0", "0", "0", "0", "0", "3", "151", "17428480", "1070764032", "0", "0", "0", "0", "0"}

goroutine 101 [running]:
github.com/prometheus/client_golang/prometheus.MustNewConstMetric(...)
        /opt/app-root/src/github.com/sustainable-computing-io/kepler/vendor/github.com/prometheus/client_golang/prometheus/value.go:107
github.com/sustainable-computing-io/kepler/pkg/collector.(*Collector).Collect(0xc000400710, 0xc0000fdf60?)
        /opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/collector/collector.go:315 +0x1a76
github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
        /opt/app-root/src/github.com/sustainable-computing-io/kepler/vendor/github.com/prometheus/client_golang/prometheus/registry.go:446 +0xfb
created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather
        /opt/app-root/src/github.com/sustainable-computing-io/kepler/vendor/github.com/prometheus/client_golang/prometheus/registry.go:538 +0xb0b

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

sunya-ch, 2022-09-05:
@jichenjc Thank you for reporting issue.
Would you mind sharing the head part from the kepler-exporter log?
This may help scope down and debug the missing labels.

```bash
2022/09/05 07:30:23 InitSliceHandler: &{map[] /sys/fs/cgroup/system.slice /sys/fs/cgroup/system.slice /sys/fs/cgroup/system.slice}
use sysfs to obtain power
2022/09/05 07:30:23 Available counter metrics: [cpu_cycles cpu_instr cache_miss]
2022/09/05 07:30:23 Available cgroup metrics: [cgroupfs_system_cpu_usage_us cgroupfs_user_cpu_usage_us cgroupfs_memory_usage_bytes cgroupfs_cpu_usage_us]
2022/09/05 07:30:23 Available kubelet metrics: [container_cpu_usage_seconds_total container_memory_working_set_bytes]
2022/09/05 07:30:23 set coreMetricIndex = 1
2022/09/05 07:30:23 set generalMetricIndex = 1
2022/09/05 07:30:23 set dramMetricIndex = 3
config EnabledEBPFCgroupID enabled:  true
```

jichenjc, 2022-09-05:
thanks for comments, here's start of log

```
# kubectl logs kepler-exporter-6qnsf -n monitoring |head
2022/09/05 06:34:10 InitSliceHandler: &{map[] /sys/fs/cgroup/cpu /sys/fs/cgroup/memory /sys/fs/cgroup/blkio}
cpu architecture Haswell, dram in GB 31
use power estimate to obtain power
2022/09/05 06:34:10 Available counter metrics: [cpu_cycles cpu_instr cache_miss]
2022/09/05 06:34:10 Available cgroup metrics: []
2022/09/05 06:34:10 Available kubelet metrics: [container_cpu_usage_seconds_total container_memory_working_set_bytes]
2022/09/05 06:34:10 set coreMetricIndex = 1
2022/09/05 06:34:10 set generalMetricIndex = 1
2022/09/05 06:34:10 set dramMetricIndex = 3
config EnabledEBPFCgroupID enabled:  true

```

worthy note is seems the pod become Running after failure above..
not sure what pod/actions did the magic as I didn't do anything..

```
# kubectl get pods -n monitoring -o wide  | grep kepler-exporter
kepler-exporter-6qnsf                  1/1     Running   2 (79m ago)   89m   172.18.0.2    kind-control-plane   <none>           <none>
```

jichenjc, 2022-09-05:
`curl localhost:9102/metrics` run this will cause the pod to crash as reported above ...

sunya-ch, 2022-09-05:
Thank you for proving the log and reporting this issue.
The exported labels is 20 if no frequency option enable in BPF (`-DCPU_FREQ`).
My guess is it is a bug that when determining the prometheus label `EnableCPUFreq` is true before attacher can check the failure. 
Could you try grepping `-DCPU_FREQ` from kepler log to confirm this assumption?

jichenjc, 2022-09-06:
thanks for the tip, yes, something wrong here,
I don't know what happened here , need dig more as brand new here :)

```
kubectl logs kepler-exporter-6qnsf -n monitoring | grep DCPU_FREQ
failed to attach perf module with options [-DNUM_CPUS=8 -DCPU_FREQ]: failed to load sched_switch: error loading BPF program: permission denied

```

sunya-ch, 2022-09-06:
@jichenjc Thank you for confirming this. 
Hope it can be fixed by this PR: https://github.com/sustainable-computing-io/kepler/pull/135

sunya-ch, 2022-09-06:
Also, it seems the BPF cannot be attached to your system even with only `DNUM_CPUS` option due to `permission denied` at loading. 
Could you provide the VM OS and kernel version?

sunya-ch, 2022-09-06:
I think the current version of kepler is not compatible with BPF library on Ubuntu yet. 
One of our team is now working on that. 

If it becomes available, BPF should be able to be attached and track CPU time of the process.
Running on top of VM would affect the visibility of physical file system and some data may not be available such as  RAPL power, CPU frequency, and hardware counters.

jichenjc, 2022-09-06:
>>>I think the current version of kepler is not compatible with BPF library on Ubuntu yet.
One of our team is now working on that.

I might miss this info, so I have to run on RHEL ? I will give a try

PR #135 (2022-09-06): set pod prometheus label after attachment
This PR set podEnergyLabels after calling AttachBPFAssets by attacher to support the case that `attacher.EnableCPUFreq` is changed to _false_ due to attachment failure.

fix #131
closed 2022-09-06 by rootfs