sustainable-computing-io/kepler #132: Same eCore being reported for three cores with different frequencies

Feelas, 2022-09-05:
**Describe the discussion topic**
Something I have noticed today when running another bunch of tests. Let's say I have three different pods, each scheduled on a CPU with a different frequency setup. What I can see in logs is similar to this (grepped to contain the barest minimum):
```
--
energy from pod (12 processes): name: fastest-pod namespace: some-namespace
        cgrouppid: 4295012128 pid: 681667144608761 comm: some-process
        ePkg (mJ): 38117 (2554154) (eCore: 35297 (2369735) eDram: 2820 (184419) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
--
energy from pod (12 processes): name: normal-pod namespace: some-namespace
        cgrouppid: 4295011720 pid: 681229057944467 comm: some-process
        ePkg (mJ): 37950 (2536826) (eCore: 35297 (2369735) eDram: 2653 (167091) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
--
energy from pod (12 processes): name: slowest-pod namespace: some-namespace
        cgrouppid: 4295011924 pid: 681413741538238 comm: some-process
        ePkg (mJ): 40101 (2657345) (eCore: 35297 (2369735) eDram: 4804 (287610) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
```
**Since the fastest pod is using using 4x CPU freq of slowest-pod and about 25% more than the normal-pod, does it make sense for eCore delta to be the same value for each pod?** 

Same behaviour continues with each refresh of the metrics. At the end, I get very similar total energy for each of the pod, despite one of those has done 4x less job due to much lower frequency. I've been able to observe eCore deltas between workloads using different frequencies using previous images.

This is using Kepler image sha256:19b72e21aa1a84f16dc35a29b8ad17a8b28097b195b75a661196657dec12da90.

Is the current behaviour the correct one?

**Additional context**
As a side note (might possibly help), I'm seeing about 15-50% of previously reported energy (when compared to running on older images) with current images.

sunya-ch, 2022-09-05:
@Feelas This is very interesting result. The currently reported pod energy is simply computed by ratio of CPU cylces counters. Could you also confirm the pod usage (CPU cycles) and the node-level energy and counter (total package energy, total CPU cycles) at the same time slot?

sunya-ch, 2022-09-06:
@Feelas  I think we found the cause of issue. I mistakenly use UncoreMetric which even division to determine the core energy. 
Please find the update in this PR. 
https://github.com/sustainable-computing-io/kepler/pull/136

Feelas, 2022-09-06:
@sunya-ch thanks for prompt intervention! Manually built an exporter image from #136 and the results make more sense towards to what was observed before, at least intuitively.

rootfs, 2022-09-06:
@Feelas thank you for confirming!