sustainable-computing-io/kepler #147: Kepler on OpenShift, energy estimates are all zeros

sallyom, 2022-09-08:
I deployed with the following commands on OpenShift 
(from PR #128 manifests/openshift/README)

```shell
oc apply --kustomize $(pwd)/manifests/openshift/cluster-prereqs
# The cluster-prereqs modifies all nodes to enable cgroupsv2. This takes a long time
# Each node is decommissioned and rebooted - may take ~20 minutes.

# Check before proceeding that all nodes are Ready and Schedulable
oc get nodes

oc apply --kustomize $(pwd)/manifests/openshift/kepler
# Check that kepler pods are up and running before proceeding

# The following script applies the kustomize files in $(pwd)/manifests/openshift/dashboard
$(pwd)/deploy-grafana.sh
```

All resources are healthy, but energy estimates are all showing zeros, like so
[image]

rootfs, 2022-09-08:
```console
$ kubectl logs -n monitoring kepler-exporter-7pzkv |head -50
2022/09/08 13:40:54 InitSliceHandler: &{map[] /sys/fs/cgroup/kubepods.slice /sys/fs/cgroup/kubepods.slice /sys/fs/cgroup/kubepods.slice}
cpu architecture Broadwell, dram in GB 15
use power estimate to obtain power
2022/09/08 13:40:54 Available counter metrics: [cpu_cycles cpu_instr cache_miss]
2022/09/08 13:40:54 Available cgroup metrics: []
2022/09/08 13:40:54 Available kubelet metrics: [container_cpu_usage_seconds_total container_memory_working_set_bytes]
2022/09/08 13:40:54 set coreMetricIndex = 1
2022/09/08 13:40:54 set generalMetricIndex = 1
2022/09/08 13:40:54 set dramMetricIndex = 3
config EnabledEBPFCgroupID enabled:  true
config getKernelVersion:  4.18
config set EnabledEBPFCgroupID to  true
perf_event_open: No such file or directory
failed to attach perf event cpu_cycles: <nil>
failed to attach perf event cpu_instr: <nil>
perf_event_open: No such file or directory
perf_event_open: No such file or directory
failed to attach perf event cache_miss: <nil>
```

sunya-ch, 2022-09-09:
just implement here.
https://github.com/sustainable-computing-io/kepler/pull/150

sallyom, 2022-10-05:
this issue is resolved with #150 AFAICT

closed 2022-10-05 by sallyom