sustainable-computing-io/kepler #153: docker(containerd): not able to read container, plan to support?

jichenjc, 2022-09-09:
**Describe the bug**
A clear and concise description of what the bug is.

https://github.com/sustainable-computing-io/kepler/blob/main/pkg/pod_lister/resolve_container.go#L202

expecting pod/crio 
this works on RHEL 

```
1:name=systemd:/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-podb4e8e0dc_caf3_4a43_bf50_f84ef5ef7a8f.slice/crio-conmon-927cffdca4f374600fe49a62bfd7f2e73ea6d37cdafb1bf8f89afb64d7e6c6a5.scope
```

but I run this on docker(containerd) got following 

```
# cat /proc/1959/cgroup
0::/system.slice/containerd.service

# crictl ps
CONTAINER           IMAGE               CREATED             STATE               NAME                      ATTEMPT             POD ID              POD
db90aabe3ba00       295c7be079025       About an hour ago   Running             nginx                     0                   6bdfe6263b2e1       nginx-deployment-6595874d85-h7csj
677824b3d4028       295c7be079025       About an hour ago   Running             nginx                     0                   caf965c196f45       nginx-deployment-6595874d85-wdtm6

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jichenjc, 2022-09-09:
I am running cgroupv1 but I think it's not related?

jichenjc, 2022-09-09:
I think this might be the way to get containerd ID ....

```
root@kind-control-plane:/proc/1959# cat cmdline
/usr/local/bin/containerd-shim-runc-v2-namespacek8s.io-id6bdfe6263b2e1e5e96192d15f365216830ba3f28b982e0bccec61b41beb7f831-address/run/containerd/containerd.sockroot@kind-control-plane:/proc/1959# crictl ps | grep 6bdfe6263b2e1e
root@kind-control-plane:/proc/1959# crictl ps | grep 6bdfe6263b2
db90aabe3ba00       295c7be079025       About an hour ago   Running             nginx                     0                   6bdfe6263b2e1       nginx-deployment-6595874d85-h7csj

```

rootfs, 2022-09-09:
@jichenjc would #152 help?

jichenjc, 2022-09-12:
I guess not .. both use containerd/docker below

RHEL 
```
[root@ji4 ~]# cat /proc/28069/cgroup
11:blkio:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/system.slice/containerd.service
10:cpuset:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope
9:memory:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/system.slice/containerd.service
8:perf_event:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope
7:devices:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/system.slice/containerd.service
6:hugetlb:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope
5:cpuacct,cpu:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/system.slice/containerd.service
4:net_prio,net_cls:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope
3:freezer:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope
2:pids:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/system.slice/containerd.service
1:name=systemd:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/system.slice/containerd.service

```

Ubuntu 
```
# cat /proc/15945/cgroup
0::/system.slice/containerd.service

root@jitest39:~# cat /proc/15945/cmdline
/usr/bin/containerd-shim-runc-v2-namespacemoby-id3f3488e110e93031b1528ecfb2bc5332eae24cf8e578b38e660755c894768196-address/run/containerd/containerd.sock
```

marceloamaral, 2022-09-12:
Is the container in a pod? The path you shared has no pod information.

I'm asking this because most of the logic was built for a path that contains the pod information...

jichenjc, 2022-09-12:
ok, I used docker at above output , seems I should use k8s ,then I used such at Ubuntu host

```
root@kind-control-plane:/# crictl info |jq '.config.containerd.runtimes'
{
  "runc": {
    "runtimeType": "io.containerd.runc.v2",
    "runtimePath": "",
    "runtimeEngine": "",
    "PodAnnotations": null,
    "ContainerAnnotations": null,
    "runtimeRoot": "",
    "options": {
      "SystemdCgroup": true
    },
    "privileged_without_host_devices": false,
    "baseRuntimeSpec": "/etc/containerd/cri-base.json",
    "cniConfDir": "",
    "cniMaxConfNum": 0
  },
  "test-handler": {
    "runtimeType": "io.containerd.runc.v2",
    "runtimePath": "",
    "runtimeEngine": "",
    "PodAnnotations": null,
    "ContainerAnnotations": null,
    "runtimeRoot": "",
    "options": {
      "SystemdCgroup": true
    },
    "privileged_without_host_devices": false,
    "baseRuntimeSpec": "/etc/containerd/cri-base.json",
    "cniConfDir": "",
    "cniMaxConfNum": 0
  }
}
root@kind-control-plane:/# crictl inspect db90aabe3ba00 |jq '.info.pid'|xargs -I {} cat /proc/{}/cgroup
0::/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-besteffort.slice/kubelet-kubepods-besteffort-pod36f20d9d_cbc1_4ebd_b111_536eaa6a332e.slice/cri-containerd-db90aabe3ba00bab92a9bd3f0b4a9face4601651c91d28c02a953a8c81ce2cc4.scope
```

```
# crictl ps
CONTAINER           IMAGE               CREATED             STATE               NAME                      ATTEMPT             POD ID              POD
db90aabe3ba00       295c7be079025       3 days ago          Running             nginx                     0                   6bdfe6263b2e1       nginx-deployment-6595874d85-h7csj
677824b3d4028       295c7be079025       3 days ago          Running             nginx                     0                   caf965c196f45       nginx-deployment-6595874d85-wdtm6
c9ccfaeab8d2c       4c1e997385b8f       3 days ago          Running             local-path-provisioner    0                   00a0f1acb89cc       local-path-provisioner-9cd9bd544-whx5v
a0abc2de4709e       a4ca41631cc7a       3 days ago          Running             coredns                   0                   1707c426eaf31       coredns-6d4b75cb6d-5kzhk
3cb62afd253ed       a4ca41631cc7a       3 days ago          Running             coredns                   0                   5fc632cd15576       coredns-6d4b75cb6d-8sfzh
dc39c4066a25e       6fb66cd78abfe       3 days ago          Running             kindnet-cni               0                   2f1bb833036cf       kindnet-2g7jj
10f1959548023       6960c0e47829d       3 days ago          Running             kube-proxy                0                   5ce6ee6f8d893       kube-proxy-7qcfd
9449d49b96f8f       aebe758cef4cd       3 days ago          Running             etcd                      0                   ec5f3b762fdb5       etcd-kind-control-plane
59b734a871285       9ef4b1de3be49       3 days ago          Running             kube-apiserver            0                   d7388bcc51046       kube-apiserver-kind-control-plane
9cd14730bbfdc       efa8a439d1460       3 days ago          Running             kube-controller-manager   0                   2617c809f1f39       kube-controller-manager-kind-control-plane
eeab955c1cbf7       41f5241e3396e       3 days ago          Running             kube-scheduler            0                   5f201df358cd7       kube-scheduler-kind-control-plane
```

jichenjc, 2022-09-12:
so looks like it should support by existing code, will add some UT to prove

jichenjc, 2022-09-13:
I got this from RHEL , same to yours @rootfs ,so we need support both (maybe more like SLES but can be done later)
```
# crictl inspect ecbcea5cd29af |jq '.info.pid'|xargs -I {} cat /proc/{}/cgroup
11:blkio:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope
10:cpuset:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope
9:memory:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope
8:perf_event:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope
7:devices:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope
6:hugetlb:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope
5:cpuacct,cpu:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope
4:net_prio,net_cls:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope
3:freezer:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope
2:pids:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope
1:name=systemd:/system.slice/docker-c27755f0fa91e81ababc85ef05cb227227a4228da2e5cb2f4999299c89d4ac69.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-burstable.slice/kubelet-kubepods-burstable-podd8992f589d8dd12c4342376ccb459375.slice/cri-containerd-ecbcea5cd29afb25ba519715e827cda9e66cd0a914207f49ce0a292a6aa84d66.scope

```

rootfs, 2022-09-13:
@jichenjc it you want to give it a try? would you also like to write a unit test too? That'll make sure the regex works for all.

jichenjc, 2022-09-13:
yes, I am working this now

marceloamaral, 2023-03-31:
@jichenjc can we close this issue?

jichenjc, 2023-04-01:
thanks for the reminder @marceloamaral

closed 2023-04-01 by jichenjc