sustainable-computing-io/kepler #154: running UT at /kepler/pkg/pod_lister always Success

jichenjc, 2022-09-09:
**Describe the bug**
A clear and concise description of what the bug is.

no matter condition it is , always success due to 
```
failed to get response: failed to read from "/var/run/secrets/kubernetes.io/serviceaccount/token": open /var/run/secrets/kubernetes.io/serviceaccount/token: no such file or directorytesting: warning: no tests to run
PASS
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jichenjc, 2022-09-09:
root cause is following function will already run at init so it will httpGet and failed with above error

func init() {
        updateListPodCache("", false)
}

jichenjc, 2022-09-09:
/assign

sunya-ch, 2022-09-09:
I think there is no unit test case applied for pod lister yet.
You should add RegisterFailHandler and RunSpecs to make it as the example in suite_test.go in cgroup/collecter/model.
https://github.com/sustainable-computing-io/kepler/blob/c65de66ea9325920561f02c97cb68934d5d29fef/pkg/cgroup/suite_test.go#L27

more information about ginkgo test here: https://onsi.github.io/ginkgo/

The init function will just log that the token path is not available. It should not make the test pass or failed except you put it in the test case.

jichenjc, 2022-09-09:
I put a very simple test in the folder 

```
(base) root@jitest19:~/kepler/pkg/pod_lister# go test ./...
ok      github.com/sustainable-computing-io/kepler/pkg/pod_lister       (cached)
(base) root@jitest19:~/kepler/pkg/pod_lister# cat pod_lister_test.go
package pod_lister

import (

        . "github.com/onsi/ginkgo"
        . "github.com/onsi/gomega"
)

var _ = Describe("Bdfasdyte order test", func() {
        It("Byte ordeafdsasdfr test", func() {
                a := "111c"
                Expect(a).To(Equal("c"))
        })
})
(base) root@jitest19:~/kepler/pkg/pod_lister#

```

and it apparently is not correct

jichenjc, 2022-09-09:
ok, let me try this

jichenjc, 2022-09-09:
indeed, sorry for confusion

closed 2022-09-09 by jichenjc

sunya-ch, 2022-09-09:
Good to hear that it's solved :)