sustainable-computing-io/kepler #160: add badges in readme

SamYuan1990, 2022-09-11:
**Is your feature request related to a problem? Please describe.**
N/A

**Describe the solution you'd like**

- [ ] go doc
- [ ] go test coverage
- [ ] CI

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

xiaopeng-zh, 2022-09-15:
@SamYuan1990 Can I take this one?

SamYuan1990, 2022-09-15:
> @SamYuan1990 Can I take this one?

welcome

xiaopeng-zh, 2022-09-16:
There are two ways to set the `go test coverage` badge.

- Static way: Just hardcode the percentage of coverage but needs to update the data when the value changes.
- Dynamic way: Get the latest coverage data & update it automatically. Ref: [go-coverage-badge](https://github.com/marketplace/actions/go-coverage-badge)

@SamYuan1990 Which way is suitable for us now?

SamYuan1990, 2022-09-16:
> There are two ways to set the `go test coverage` badge.
>
> * Static way: Just hardcode the percentage of coverage but needs to update the data when the value changes.
> * Dynamic way: Get the latest [...]

I suppose if the dynamic way means the test coverage will updated once after each PR merged, then I prefer dynamic way.  
we should not update data for test coverage in manual. 

@rootfs , any comments?

rootfs, 2022-09-16:
dynamic way it is, thank you @xiaopeng-zh for taking this!

xiaopeng-zh, 2022-09-19:
Ref: https://github.com/sustainable-computing-io/kepler/pull/189, the PR merged, this can be closed.

closed 2022-09-19 by rootfs