sustainable-computing-io/kepler #163: Possible wrong spelling on variable name

williamcaban, 2022-09-11:
It looks to me the `NODE_ENERGY_STAT_METRRIC = "node_energy_stat"` on line29 https://github.com/sustainable-computing-io/kepler/blob/5d4f4e93c8a28c2e21244ea1dfa4817affc7bb44/pkg/collector/collector.go#L29 might be spelling error. 

Can anyone confirm? The same spelling is also used on the following files:

https://github.com/sustainable-computing-io/kepler/blob/424d2f88d2c050a354c023e8b830afefd312baf4/pkg/collector/collector.go#L123

https://github.com/sustainable-computing-io/kepler/blob/fb8af8c979516e4015b2dbebef88d1896e613e03/pkg/collector/collector_test.go#L133

rootfs, 2022-09-11:
@williamcaban nice code reading! PR welcome

marceloamaral, 2023-05-18:
@williamcaban can you create a PR for this?

closed 2023-08-28 by stale[bot]