sustainable-computing-io/kepler #173: Kepler on s390 platform

mengfj, 2022-09-13:
**Is your feature request related to a problem? Please describe.**
As an s390 application developer or SRE, I'd like to get the energy consumption measurement on x390, so that I can have real-time monitoring of power consumption of my environments on s390. 

**Describe the solution you'd like**
I would like to have Kepler running on s390x platform and measure the power consumption. 

**Describe alternatives you've considered**
Currently, there is no alternative on s390. 

**Additional context**
Need check if there is any compatibility issue of Kepler running on s390x platform and fix it if any.

mengfj, 2022-09-13:
@rootfs please help assign this to myself, @jichenjc and other colleagues. We'll start from the verification. Thanks!

huoqifeng, 2022-09-15:
https://github.com/sustainable-computing-io/kepler/issues/184