sustainable-computing-io/kepler #179: getKernelVersion doesn't work at all

jichenjc, 2022-09-14:
**Describe the bug**
A clear and concise description of what the bug is.

https://github.com/sustainable-computing-io/kepler/blob/main/pkg/config/config.go#L65

paste those to https://go.dev/play/ and run it 
```
package main

import (
	"encoding/json"
	"fmt"

	"github.com/zcalusic/sysinfo"
)

func main() {

	var si sysinfo.SysInfo

	si.GetSysInfo()

	data, err := json.MarshalIndent(&si, "", "  ")
	if err == nil {
		var result map[string]map[string]string
		if err = json.Unmarshal(data, &result); err != nil {
			fmt.Println("----")
			fmt.Println(err)
			fmt.Println("----")
		}
	}
	fmt.Println("done")
}

```

```
----
json: cannot unmarshal number into Go value of type string
----
done

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jichenjc, 2022-09-14:
I will submit a PR to fix this soon

jichenjc, 2022-09-14:
https://github.com/zcalusic/sysinfo/blob/99e836ba64f229922382eecd8a346e6b3d1b560a/kernel_linux.go#L22

we can refer to this function and just use it instead of using the whole library

SamYuan1990, 2022-09-14:
but `fmt.Println(si.Kernel.Version)` is empty at go playground.

jichenjc, 2022-09-14:
I don't think we should use this library (at least for now) at all
as you can see , they just read a file and we should get from that file directly
https://github.com/zcalusic/sysinfo/blob/99e836ba64f229922382eecd8a346e6b3d1b560a/kernel_linux.go#L22

SamYuan1990, 2022-09-14:
> I don't think we should use this library (at least for now) at all as you can see , they just read a file and we should get from that file directly https://github.com/zcalusic/sysinfo/blob/99e836ba64f229922382eecd8a346e6b3d1b560a/kernel_linux.go#L22

I am not sure if this one looks better 
https://github.com/cilium/ebpf/blob/b81c37125bd70a4f086b073e821a5ea79b31e394/features/version.go#L12
...// ignore for code trace stack
https://github.com/cilium/ebpf/blob/3418f341ac0460cf4164146e0c1ec6c40156b4fc/internal/vdso.go#L21

marceloamaral, 2022-09-14:
We can read the os-release files instead of use the library.
For example [docker-slim](https://github.com/docker-slim/docker-slim/blob/10c72982ce4c6e4bb8f76ffc2a50faeb70c6a2f0/pkg/system/os_release.go#L14) reads different files depending on the OS distribution. 
Note that we also need to support macOS for developing purpose.

Could you please submit a PR for this?

rootfs, 2022-09-14:
we can read `/proc/version` 
```go
package main

import (
	"fmt"
	"log"
	"os"
)

func main() {
	output, err := os.ReadFile("/proc/version")
	if err != nil {
		log.Fatal(err)
	}

	fmt.Printf("output: %q\n", string(output))
}
```
[Here](https://go.dev/play/p/YULURUIhHP9) is the go  playground test

rootfs, 2022-09-14:
`GetSysInfo` gets kernel version from [/proc/sys](https://github.com/zcalusic/sysinfo/blob/99e836ba64f229922382eecd8a346e6b3d1b560a/kernel_linux.go#L22)

While [uname](https://github.com/wertarbyte/coreutils/blob/master/src/uname.c#L287) and golang [Uname](https://pkg.go.dev/syscall#Uname) use syscall. 

Let's switch to the uname way.

jichenjc, 2022-09-15:
yes, thanks for the info, I will update the code with your comment soon

BTW I don't have Mac env, guess the unix way should be able to cover that as well?

marceloamaral, 2022-09-15:
Here is what I get:

```bash
uname -a
Darwin MacBook-Pro-3.local 21.6.0 Darwin Kernel Version 21.6.0: Sat Jun 18 17:07:25 PDT 2022; root:xnu-8020.140.41~1/RELEASE_X86_64 x86_64
```

```bash
name -v
Darwin Kernel Version 21.6.0: Sat Jun 18 17:07:25 PDT 2022; root:xnu-8020.140.41~1/RELEASE_X86_64
```

jichenjc, 2022-09-15:
is https://github.com/sustainable-computing-io/kepler/issues/179#issuecomment-1247090703 works for you?

anyway, as metioned, previously the mac version from sysinfo library doesn't have Mac implementation at all...

jichenjc, 2022-09-21:
I only tested the unix way in golang locally, but found those in latest build env

```
Not able to parse kernel version, use -1 instead
config getKernelVersion:  -1
Not able to parse kernel version, use -1 instead

```
need check why ...

jichenjc, 2022-09-21:
5.4.0-48-generic is my OS, so based on current logic, it's 5.4. (not the .)
so not able to parse the kernel version

PR #187 (2022-09-16): detect kernel version through unix way
Fixes #179
closed 2022-09-26 by jichenjc