sustainable-computing-io/kepler #182: microshift as an environment for integration test

SamYuan1990, 2022-09-14:
**Is your feature request related to a problem? Please describe.**
add microshift as micro of openshift as integration test env

**Describe the solution you'd like**
considering github action providing agent as ubuntu, should try command below to see if we are able to use microshift as a k8s cluster for integration test.
https://github.com/thinkahead/microshift/blob/main/install-ubuntu22.04.sh

**Describe alternatives you've considered**
n/A

**Additional context**
n/A

rootfs, 2022-09-14:
cc @sallyom

SamYuan1990, 2022-09-15:
@rootfs ,@sallyom , may I know
- when will this think a head back to https://github.com/openshift/microshift
- is there any sample to upload local docker image to microshift so that in PR we can upload PR build to microshift for test?

cooktheryan, 2022-09-26:
I do know that @sallyom was attempting to begin the work on integrating the OCP CI framework into this repository. She may also be able to comment if the ability still exists to run MicroShift completely containerized which would allow for us to successfully use it with Ubuntu. If not we may be able to fork the MicroShift testing framework for this project as well to deploy microshift and then the Kepler objects

sallyom, 2022-09-26:
the manifests/openshift are working now AFAIK, see here https://github.com/sustainable-computing-io/kepler/blob/main/manifests/openshift/README.md  

I will report back wrt running on MicroShift - haven't tried that - however, if you have an OpenShift cluster you can deploy with the above README

SamYuan1990, 2022-09-27:
> the manifests/openshift are working now AFAIK, see here https://github.com/sustainable-computing-io/kepler/blob/main/manifests/openshift/README.md
> 
> I will report back wrt running on MicroShift - haven't tried that - however, if you have an OpenShift cluster you can deploy with the above README

do you know if we are able to make OpenShift running in github action(github hosted agent) ?

SamYuan1990, 2022-12-20:
@sallyom / @rootfs ,  would you like ref to https://github.com/sustainable-computing-io/KeplerK8SAction/pull/8 implements microshift support in our customer github action? then we can reuse microshift as k8s cluster in our CI for all projects. (kepler, operator, helm etc...)

SamYuan1990, 2023-05-17:
@rootfs ,@sallyom  any updates for this one?
are we able to enable microshift in github action?

marceloamaral, 2023-05-18:
any update on this?

SamYuan1990, 2023-06-05:
> Not sure about the GitHub action for Microshift but I have recently used Microshift for CI purposes and I use the default github-runners and run Microshift in a container as I had an issue in the past deploying Micorshift straight on Ubuntu and there is an [open issue ](https://github.com/openshift/microshift/issues/984)as well related to this. If this doesn't work then we can definitely explore the container option as well. In my experience, it works fine. Just have some extra steps😅

yeah... as https://github.com/sustainable-computing-io/kepler/issues/182#issuecomment-1259524816
hence I suppose currently some of major question:
1. are we going to handle this ticket in 0.6 release?
2. are we need make a further design for this?
assign to @rootfs, and let's wait further information.

rootfs, 2023-07-26:
this is finished by @vprashar2929

closed 2023-07-26 by rootfs