sustainable-computing-io/kepler #184: Build s390x image via corss-platform build

huoqifeng, 2022-09-15:
Leverage exiting docker buildx cross-platform to build s390x image.

cooktheryan, 2022-09-15:
this could be a cool tool to leverage for this work https://github.com/docker/build-push-action

Example https://github.com/migtools/gitops-primer/blob/main/.github/workflows/validate-primer.yaml#L361-L368

huoqifeng, 2022-09-16:
I noticed package `nvidia-driver-NVML-515.48.07-1.el8.x86_64.rpm` here in base image https://github.com/sustainable-computing-io/kepler/blob/main/build/Dockerfile.base#L6, looks only has x86 arch supported?

huoqifeng, 2022-09-16:
@mengfj @SamYuan1990 @jichenjc I don't think LinuxONE has nvidia card installed, so it's reasonable that package `nvidia-driver-NVML` does not support s390x. Is this package key to kepler?

SamYuan1990, 2022-09-16:
> @mengfj @SamYuan1990 @jichenjc I don't think LinuxONE has nvidia card installed, so it's reasonable that package `nvidia-driver-NVML` does not support s390x. Is this package key to kepler?

Personally,  I suppose we need to have a MVP on s390x.

Specifically for automation point of view, the MVP is to build s390x based golang binary and running it in a s390x based fedora image?
As far as I know:
- there lots of golang build tag in current build, sooner or later we should have one for s390x. (as so far kepler build logic is basing on build tag for different hardware.)
- then for docker image, one thing hard is for base image, others is looking for different libs, as nvidia one found today.
- we can use matric for x86/arm/s390x to control the build.
ideally, the make script will look like either
```
make ${arch}
```
or
``` 
# in make file
if $uname ... -eq s390x 
GO_BUILD_TAGS = $GO_BUILD_TAGS + s390_tag
fi
```

personally, consider we may have different docker file for different arch, I prefer below
```
make ${arch}
```

@rootfs , some questions:
1. for any vm without nvidia hardware? what will be happen? what build tag should we use?
2. any other comments?

rootfs, 2022-09-16:
@huoqifeng I don't know much about nvidia on s390, the nvml pkg can be [turned on and off](https://github.com/sustainable-computing-io/kepler/blob/main/cmd/exporter.go#L39) at runtime. You can make a conditional build for s390 to turn off nvml support if necessary.

closed 2023-05-24 by stale[bot]