sustainable-computing-io/kepler #192: CI needs more instrumentation

rootfs, 2022-09-17:
**Describe the bug**
CI is getting flaky
```
+ ./hack/cluster-deploy.sh
waiting for cluster-clean to finish
+ source cluster-up/common.sh
++ set -e
++ '[' kubernetes = kind ']'
Deploying manifests...
+ CLUSTER_PROVIDER=kubernetes
+ MANIFESTS_OUT_DIR=_output/manifests/kubernetes/generated
+ main pipefail
+ '[' '!' -d _output/manifests/kubernetes/generated ']'
+ echo 'Deploying manifests...'
+ kubectl apply -f _output/manifests/kubernetes/generated
namespace/kepler created
clusterrole.rbac.authorization.k8s.io/kepler-clusterrole created
clusterrolebinding.rbac.authorization.k8s.io/kepler-clusterrole-binding created
serviceaccount/kepler-sa created
daemonset.apps/kepler-exporter created
service/kepler-exporter created
servicemonitor.monitoring.coreos.com/kepler-exporter created
+ kubectl rollout status daemonset kepler-exporter -n kepler --timeout 60s
Waiting for daemon set "kepler-exporter" rollout to finish: 0 of 1 updated pods are available...
error: timed out waiting for the condition
make: *** [Makefile:165: cluster-sync] Error 1
Error: Process completed with exit code 2.
```
**To Reproduce**
Recent CI results [here](https://github.com/sustainable-computing-io/kepler/actions/runs/3071410017/jobs/4962047170#step:8:237)
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

SamYuan1990, 2022-09-17:
https://github.com/sustainable-computing-io/kepler/actions/runs/3073381944/jobs/4965474260#step:8:233
I see latest CI passed, @rootfs  is this issue able to reproduce easily?

rootfs, 2022-09-17:
let's keep an eye on it and re-open this issue if it happens again.

marceloamaral, 2022-09-21:
Maybe we need a larger timeout?
Let's see if it happen again

rootfs, 2022-09-22:
https://github.com/sustainable-computing-io/kepler/pull/231#issuecomment-1254422128

marceloamaral, 2022-09-23:
Thanks @SamYuan1990!

What we need is the artifacts https://docs.github.com/en/actions/using-workflows/storing-workflow-data-as-artifacts

So we will dump the logs and description of the pods to files in the artifact folder
Something like:
```
kubectl describe pod -n $1 $2 > $artifacts/describe/pod/$2
kubectl describe logs -n $1 $2 > $artifacts/logs/pod/$2
```

But, to save space we just dump the info when something fail

SamYuan1990, 2022-09-24:
> Thanks @SamYuan1990!
> 
> What we need is the artifacts https://docs.github.com/en/actions/using-workflows/storing-workflow-data-as-artifacts
> 
> So we will dump the logs and description of the pods to files in the artifact folder Something like:
> 
> ```
> kubectl describe pod -n $1 $2 > $artifacts/describe/pod/$2
> kubectl logs -n $1 $2 > $artifacts/logs/pod/$2
> ```
> 
> But, to save space we just dump the info when something fail

any suggestion for fail handle?
I suppose our points is 
```
when fail happnes.
kubectl describe pod -n $1 $2 > $artifacts/describe/pod/$2
kubectl logs -n $1 $2 > $artifacts/logs/pod/$2
```

marceloamaral, 2022-09-25:
@SamYuan1990 something like:
```
function logdump {
   kubectl get po -A -o custom-columns=namespace:.metadata.namespace,name:.metadata.name | tail +2 | while read -r line ; do 
     line=($line)
     namespace=${line[0]}
     podname=${line[1]}
     
     mkdir -p $artifacts/describe/pod/$namespace
     mkdir -p$artifacts/logs/pod/$namespace
     
     kubectl describe pod -n $namespace $podname > $artifacts/describe/pod/$namespace/$podname
     kubectl logs -n $namespace $podname > $artifacts/logs/pod/$namespace/$podname
   done
   
}

trap 'logdump' ERR
```

```

SamYuan1990, 2022-10-19:
@rootfs ,is this issue reproduced after we fixed panic issue?

rootfs, 2022-10-19:
@SamYuan1990 we are not seeing it recently.

closed 2022-10-19 by rootfs