sustainable-computing-io/kepler #193: cover: open coverage.out: no such file or directory

rootfs, 2022-09-17:
**Describe the bug**
go coverage CI error

```
--- PASS: TestPodLoader (0.00s)
PASS
coverage: 24.6% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/podlister	0.052s
?   	github.com/sustainable-computing-io/kepler/pkg/power/acpi	[no test files]
?   	github.com/sustainable-computing-io/kepler/pkg/power/gpu	[no test files]
?   	github.com/sustainable-computing-io/kepler/pkg/power/rapl	[no test files]
?   	github.com/sustainable-computing-io/kepler/pkg/power/rapl/source	[no test files]
cover: open coverage.out: no such file or directory
Error: Process completed with exit code 1.
```

**To Reproduce**
This is in this [run result](https://github.com/sustainable-computing-io/kepler/actions/runs/3071410020/jobs/4962047268)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

rootfs, 2022-09-17:
cc @xiaopeng-zh

SamYuan1990, 2022-09-17:
@rootfs , I will update ...
btw, I hope you can merge the PR in carefully .... https://github.com/sustainable-computing-io/kepler/actions/runs/3066091930/jobs/4961441566#step:6:358

closed 2022-09-17 by rootfs