sustainable-computing-io/kepler #197: Prometheus not able monitor the metrics from kepler namespace by default

kenplusplus, 2022-09-17:
**Describe the bug**
In [commit](https://github.com/sustainable-computing-io/kepler/commit/7510936f96c9e971c3a55de3abe4ea1f46a55cac), the namespace is changed from monitoring => kepler. This break prometheus discovery the metrics from the namespace monitoring by default.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the readme to deploy Kelper according to https://github.com/sustainable-computing-io/kepler/blob/main/manifests/kubernetes/deployment.yaml
2. Enable service monitoring via https://github.com/sustainable-computing-io/kepler/blob/main/manifests/kubernetes/keplerExporter-serviceMonitor.yaml
3. On promethus-k8s service web http://<prometheus_service_ip>:9090/, there is no active target found for [serviceMonitor/monitoring/kepler-exporter/0 (0 / 42 active targets)](http://10.102.132.81:9090/service-discovery?search=#serviceMonitor/monitoring/kepler-exporter/0)

**Expected behavior**
serviceMonitor should found target for kepler-exporter in prometheus Service Monitor target.

**Screenshots**
[image]

kenplusplus, 2022-09-17:
@marceloamaral I think there are two ways to fix this issue:
1. Rollback previous commit, put the namespace back to "monitoring", then it will allow promethus works without any addtional changes
2. Or, give step to customize the prometheus to monitoring different namespace like https://prometheus-operator.dev/docs/kube/monitoring-other-namespaces/

What's your thoughts?

rootfs, 2022-09-17:
Thank you for this diagnosis, the problem looks to be in the ServiceMonitor namespace [here](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/kubernetes/keplerExporter-serviceMonitor.yaml#L8), would you like to update the ns there and see if it works? @kenplusplus

kenplusplus, 2022-09-17:
@rootfs Actually, I have tried, it is not enough. Because prometheus only has permission to access the metrics from the namespace of "monitoring". If want prometheus monitor multiple namespace, you have to configure RBAC rules in promethus operators like below, see [refer](https://prometheus-operator.dev/docs/kube/monitoring-other-namespaces/). It really love the namespace kepler, then we should add some yaml to customize prometheus on this. I could try later. The simplest way is using the namespace "monitoring" which supported by prometheus by default.

local kp =
  (import 'kube-prometheus/main.libsonnet') +
  {
    values+:: {
      common+: {
        namespace: 'monitoring',
      },

      prometheus+:: {
        namespaces: ["default", "kube-system", "monitoring", "**kepler**"],
      },
    },
  };

{ 'setup/0namespace-namespace': kp.kubePrometheus.namespace } +
{
  ['setup/prometheus-operator-' + name]: kp.prometheusOperator[name]
  for name in std.filter((function(name) name != 'serviceMonitor' && name != 'prometheusRule'), std.objectFields(kp.prometheusOperator))
} +
// serviceMonitor and prometheusRule are separated so that they can be created after the CRDs are ready
{ 'prometheus-operator-serviceMonitor': kp.prometheusOperator.serviceMonitor } +
{ 'prometheus-operator-prometheusRule': kp.prometheusOperator.prometheusRule } +
{ 'kube-prometheus-prometheusRule': kp.kubePrometheus.prometheusRule } +
{ ['alertmanager-' + name]: kp.alertmanager[name] for name in std.objectFields(kp.alertmanager) } +
{ ['blackbox-exporter-' + name]: kp.blackboxExporter[name] for name in std.objectFields(kp.blackboxExporter) } +
{ ['grafana-' + name]: kp.grafana[name] for name in std.objectFields(kp.grafana) } +
{ ['kube-state-metrics-' + name]: kp.kubeStateMetrics[name] for name in std.objectFields(kp.kubeStateMetrics) } +
{ ['kubernetes-' + name]: kp.kubernetesControlPlane[name] for name in std.objectFields(kp.kubernetesControlPlane) }
{ ['node-exporter-' + name]: kp.nodeExporter[name] for name in std.objectFields(kp.nodeExporter) } +
{ ['prometheus-' + name]: kp.prometheus[name] for name in std.objectFields(kp.prometheus) } +
{ ['prometheus-adapter-' + name]: kp.prometheusAdapter[name] for name in std.objectFields(kp.prometheusAdapter) }

rootfs, 2022-09-17:
@sallyom updated the ns for openshift and it is working there, is there any special openshift that we can use for kubernetes?

rootfs, 2022-09-17:
@kenplusplus enabling monitoring other ns looks the way to go per the [openshift config](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/openshift/dashboard/00-openshift-monitoring-user-projects.yaml)

kenplusplus, 2022-09-17:
> @kenplusplus enabling monitoring other ns looks the way to go per the [openshift config](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/openshift/dashboard/00-openshift-monitoring-user-projects.yaml)

Yes, I just saw that~~ openshift specific. I am thinking to use [promethus-operator](https://github.com/prometheus-operator/prometheus-operator/blob/main/Documentation/user-guides/getting-started.md) instead of kube-promethus because configure RBAC for new namespace is too complex.

commit 9738ffa5 (2022-09-18): prometheus: Fixes #197 permission issue for service monitoring

1. Move the service monitoring to the namespace of kepler align with deployment
2. Add Role and RoleBinding to allow prometheus monitoring the namespace kepler

This has been tested on a local kubernetes cluster.
commit dcf11173 (2022-09-18): Merge pull request #200 from kenplusplus/fix-namespace-rbac

prometheus: Fixes #197 permission issue for service monitoring
closed 2022-09-18 by rootfs