sustainable-computing-io/kepler #206: github-action user can't push commit back

SamYuan1990, 2022-09-19:
**Describe the bug**
github-action user can't push commit back

**To Reproduce**
Steps to reproduce the behavior:
1. See error https://github.com/sustainable-computing-io/kepler/actions/runs/3079415202/jobs/4980117646#step:9:7

**Expected behavior**
github-action can push commit back as sample https://github.com/SamYuan1990/kepler/commit/62d342d974d9681ff94e4537e851174f3fce7b4d

**Screenshots**
n/A

**Desktop (please complete the following information):**
n/A

**Smartphone (please complete the following information):**
n/A

**Additional context**
n/A

SamYuan1990, 2022-09-19:
[image]
local fork setting is empty

rootfs, 2022-09-20:
finally, we had our [first bot commit](https://github.com/sustainable-computing-io/kepler/commit/4af2a4a3357833f047aac637dfcd9c1f9ce05e5d)

rootfs, 2022-09-20:
This is what I've done:
- create a bot account, create a personal access token (PAT)
- add the bot account into repo with write permission
- add bot PAT to GH secrets
- allow bot to bypass:
[image]
- add the following to GH top action:
```yaml
permissions:
  pull-requests: write
  contents: write
  repository-projects: write
```
- add the following to GH checkout action (this is very important!!!)
```yaml
- uses: actions/checkout@v2
      with:
        persist-credentials: false
        fetch-depth: 0
```
- use the bot to push changes:
```yaml
    - name: Commit changes
      if: github.event_name != 'pull_request' && steps.verify-changed-files.outputs.files_changed == 'true' 
      run: |
        git config --local user.email "info@sustainable-computing.io"
        git config --local user.name "sustainabile-computing-bot"
        git add README.md
        git commit -m "bot: Updated coverage badge."
    - name: Push changes
      if: github.event_name != 'pull_request' && steps.verify-changed-files.outputs.files_changed == 'true' 
      uses: ad-m/github-push-action@master
      with:
        github_token: ${{ secrets.GH_BOT_SECRET }}
        branch: ${{ github.head_ref }}
```

closed 2022-09-20 by rootfs