sustainable-computing-io/kepler #210: Accelerate unit_test_with_bcc

rootfs, 2022-09-19:
**Is your feature request related to a problem? Please describe.**
Accelerate [unit_test_with_bcc](https://github.com/sustainable-computing-io/kepler/blob/main/.github/workflows/unit_test.yml#L64-L83) 

**Describe the solution you'd like**
Currently the test spends much time on preparing the bcc on ubuntu. 

**Describe alternatives you've considered**
Since we already build bcc in [kepler base](https://github.com/sustainable-computing-io/kepler/blob/main/build/Dockerfile.base) container image, this could be optimized by running tests inside supported containers 

**Additional context**
@SamYuan1990 @cooktheryan

SamYuan1990, 2022-09-20:
- we have different tag used for different system.
- our code coverage, as unit test just focusing on kepler/kepler core logic but not each tag.
- for each build tag targeting with specific system/hardware, provider should keep eye on them self?
- if possible should we test with specific system/hardware? I suppose yes.
- Here comes the thing at unit test scope, as should we run the test base on a clean env or base on docker?

I want to do unit test with a clean env. which means we have to install bcc from zero.

- for container or not
as unit test in different github action with docker build, if we just use base container, which means each time, it will use lates t base image on main branch but not the pr one. otherwise, to ensure we use the base image build by pr, it's seems we need to install bcc during docker build isn't we?

as summary, I suppose accelerate with docker means we need build docker image and use it to test ...
if we install bcc during docker image, what's different with install bcc in github agent?

rootfs, 2022-09-20:
The [kepler base container](quay.io/sustainable_computing_io/kepler_base:latest) has already installed bcc. No build is needed. It is the target environment where kepler runs. Aligning the CI setup to this environment is closer to what we need.

SamYuan1990, 2022-09-20:
> The [kepler base container](quay.io/sustainable_computing_io/kepler_base:latest) has already installed bcc. No build is needed. It is the target environment where kepler runs. Aligning the CI setup to this environment is closer to what we need.

I also checked bcc https://github.com/iovisor/bcc/blob/master/INSTALL.md#ubuntu---binary and ... as ubuntu's release for bcc is still out of date...

k... let's try to use base container to run the unit test.

PR #216 (2022-09-20): try to use docker for test
resolve #210
closed 2022-09-20 by rootfs