sustainable-computing-io/kepler #211: some pod energy not reported

jichenjc, 2022-09-20:
**Describe the bug**
A clear and concise description of what the bug is.

Ubuntu env (kind running on a Ubuntu host) 

```
# cat /proc/18005/cgroup
1:cpuacct:/
0::/system.slice/docker-3f3488e110e93031b1528ecfb2bc5332eae24cf8e578b38e660755c894768196.scope/kubelet.slice/kubelet-kubepods.slice/kubelet-kubepods-podd59c0a68_38a6_49a9_8228_f58bde917e74.slice/cri-containerd-dc39c4066a25edda33a0c58d85a96cf27bbb7d3d429a9a7e3f243e9c3507c1b8.scope

but in logs

pid 18005 cgroup 0 cmd kindnetd

2022/09/20 01:30:32 failed to resolve pod for cGroup ID 0: failed to open cgroup description file for pid 18005: open /proc/18005/cgroup: no such file or directory, set podName=system_processes

there is no pod info in final output 

energy from pod (1 processes): name: local-path-provisioner-9cd9bd544-whx5v namespace: local-path-storage
        cgrouppid: 0 pid: 1891 comm: uwsgi
        ePkg (mJ): 14 (14) (eCore: 14 (14) eDram: 176 (176) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
        avgFreq: 0.00
        CPUTime:  0 (0)
        counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instr:0 (0)]
        cgroupfs: map[]
        kubelets: map[container_cpu_usage_seconds_total:0 (437) container_memory_working_set_bytes:0 (10362880)]

energy from pod (140 processes): name: system_processes namespace: system
        cgrouppid: 0 pid: 1319 comm: kcs-term
        ePkg (mJ): 14 (6023) (eCore: 14 (6023) eDram: 176 (78436) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
        avgFreq: 0.00
        CPUTime:  246000 (295147978162387)
        counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instr:0 (0)]
        cgroupfs: map[]
        kubelets: map[container_cpu_usage_seconds_total:1 (47506) container_memory_working_set_bytes:266240 (1093697536)]

energy from pod (2 processes): name: coredns-6d4b75cb6d-5kzhk namespace: kube-system
        cgrouppid: 0 pid: 2127 comm: python3
        ePkg (mJ): 14 (6023) (eCore: 14 (6023) eDram: 176 (78436) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
        avgFreq: 0.00
        CPUTime:  1005 (307397)
        counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instr:0 (0)]
        cgroupfs: map[]
        kubelets: map[container_cpu_usage_seconds_total:0 (2658) container_memory_working_set_bytes:0 (19816448)]

energy from pod (1 processes): name: kube-controller-manager-kind-control-plane namespace: kube-system
        cgrouppid: 0 pid: 688 comm: multipathd
        ePkg (mJ): 14 (6023) (eCore: 14 (6023) eDram: 176 (78436) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
        avgFreq: 0.00
        counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instr:0 (0)]
        cgroupfs: map[]
        kubelets: map[container_cpu_usage_seconds_total:0 (15764) container_memory_working_set_bytes:0 (52715520)]

node energy (mJ):
        ePkg: 53 (eCore: 53 eDram: 701 eUncore: 0) eGPU: 0 eOther: 0

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jichenjc, 2022-09-20:
>>>I suspect kindnetd is in a different pid namespace (i.e. pid 18005, seen by kepler ebpf may be different when kepler userspace runtime reads from its /proc namespace).

ok, I might need some guide on how the pod energy is actually calculated
I know we have ebpf, we have cgroup data, what's the formula to do the calcuation? 
e.g some record should be in cgroup data first then find corresponding record in ebpf? 

@rootfs

rootfs, 2022-09-20:
the current formula uses [ratio based](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/model/estimate.go#L95-L120) method. If [enabled](https://github.com/sustainable-computing-io/kepler-estimator#patch-estimator-sidecar-to-kepler-daemonset), there is also a model based [dynamic power estimate](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/model/estimate.go#L121-L167)

You'll find the energy calculation is done [here](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/reader.go#L310-L319) on the pods

rootfs, 2022-09-20:
>  some record should be in cgroup data first then find corresponding record in ebpf?

ebpf gets the info associated with the running process: cgroup id, pid, cpu runtime, perf counters, etc. This info is coupled with userspace info collected from sysfs and feed into the energy estimate formula to get the pod energy.

jichenjc, 2022-09-21:
indeed, previously I used 
BM => VM => docker (kind) => k8s 
now I switched to 
BM => VM => k8s

it's much better now, will check all the links you provided, thanks

closed 2022-09-21 by jichenjc