sustainable-computing-io/kepler #212: log/println usage

jichenjc, 2022-09-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

we use fmt.Println , log.Println etc in code

we should follow best practice to include log only 
and use different level such as Log.V(3) / V(4) etc to show different logs with different level

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

marceloamaral, 2022-09-21:
Right, this was in my TODO list...

It will be great to have different verbosity levels....

rootfs, 2022-09-25:
closed by #238

closed 2022-09-25 by rootfs