sustainable-computing-io/kepler #222: CI: don't run test post PR merge

rootfs, 2022-09-20:
**Is your feature request related to a problem? Please describe.**
CI: don't run test post PR merge

**Describe the solution you'd like**
Currently the CI runs tests and commit post PR merge, this takes time to see the merge. Since the tests are already executed when the PR is opened,  we only need to commit changes once the PR is merged

**Describe alternatives you've considered**

**Additional context**
@SamYuan1990

marceloamaral, 2022-09-21:
If it is tested pre merge, do we need to test again post merge?

SamYuan1990, 2022-09-21:
> If it is tested pre merge, do we need to test again post merge?

for unit test, we relay on it for test coverage badge.
for docker build, yes, there no unit test for image. in fact test for docker file equals with build isn't it?
for integration test ... as I said, the reason is to ensure user use this project and receive a good result.

any comments?

SamYuan1990, 2022-09-21:
@rootfs , are we good enough with necessary post action running for now?

SamYuan1990, 2022-09-21:
I suppose so far it take about 5 mins for PR as the integration test take the longest time.
for post merge, I am not sure if we can save 1 more minutes as `CTR_CMD = docker`?
I am going to take a rest. as to tired to making mistake as https://github.com/sustainable-computing-io/kepler/pull/229 ...
will check msg tomorrow to avoid making more mistake.

rootfs, 2022-09-21:
kind test takes a bit longer but overall quite an improvement
[image]

closed 2022-09-21 by rootfs