sustainable-computing-io/kepler #232: Exporting Workload Performance Metric as an independent metric that can be queried via Prometheus

wangchen615, 2022-09-22:
**Is your feature request related to a problem? Please describe.**
Currently, the `pod_energy_stat` metric is a summary object that includes all types of energy consumption metrics, resource usage metrics, and performance metrics like `curr_cpu_instr="18972163"` and `curr_cpu_time="305"`.

**Describe the solution you'd like**
We would like these performance metrics to be exported as independent queryable metrics so they can be plotted in Grafana Dashboard.

**Describe alternatives you've considered**
For Grafana visualization, we need queryable metrics.

**Additional context**
This would be useful to demonstrate how [clever](https://github.com/sustainable-computing-io/clever) recommender guarantees the workload performance when CPU frequencies are tuned down.

jichenjc, 2022-09-26:
to confirm , so you want to expose something like following 

[]string{"curr_bytes_read", "curr_bytes_writes", "curr_cache_miss", "curr_cgroupfs_cpu_usage_us", "curr_cgroupfs_memory_usage_bytes", "curr_cgroupfs_system_cpu_usage_us", "curr_cgroupfs_user_cpu_usage_us", "curr_cpu_cycles", "curr_cpu_instr", "curr_cpu_time"}

into the prometheus ?

marceloamaral, 2022-09-26:
I am working on updating all metrics to be more compliant with Prometheus metrics best practices
So, I will send a draft soon and try to solve the problems here!

marceloamaral, 2023-05-18:
We fixed this.

closed 2023-05-18 by marceloamaral