sustainable-computing-io/kepler #248: Panic when kernel header version is different from kernel version

nikimanoledaki, 2022-09-26:
**Describe the bug**
Kepler panics when it does not find kernel-headers (`linux-headers-xyz`) with the same version as the kernel version (`uname -r`). 

**To Reproduce**
My kernel version is:
```
uname -r
5.10.77
```
I cannot find kernel headers for this version! As you can see here: https://packages.ubuntu.com/focal-updates/devel/ there are only `linux-headers-5.10.0-*`, `linux-headers-5.11.0-*`  and beyond, where the patch is always `0`.

It is possible to install headers at a higher version than that of the kernel, so I installed `linux-headers-5.11.0`.

However, now the `kepler-exporter` Pod fails with a `CrashLoopBackOff`. 

Here are the logs:
```
$ k logs pod/kepler-exporter-8grrb -n kepler
I0926 15:44:14.586590       1 config.go:62] config EnabledEBPFCgroupID enabled:  true
I0926 15:44:14.586787       1 config.go:63] config getKernelVersion:  5.1
I0926 15:44:14.586828       1 config.go:67] config set EnabledEBPFCgroupID to  false
modprobe: FATAL: Module kheaders not found in directory /lib/modules/5.10.77
chdir(/lib/modules/5.10.77/build): No such file or directory
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x81ad16]

goroutine 1 [running]:
github.com/iovisor/gobpf/bcc.(*Module).Load(0x0, {0xd367ec, 0xc}, 0x2?, 0x2?, 0xc0000bb060?)
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/vendor/github.com/iovisor/gobpf/bcc/module.go:202 +0x36
github.com/iovisor/gobpf/bcc.(*Module).LoadTracepoint(...)
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/vendor/github.com/iovisor/gobpf/bcc/module.go:182
github.com/sustainable-computing-io/kepler/pkg/attacher.loadModule({0x15bf420?, 0xd35377?, 0x0?}, {0xc0000d6d60, 0x2, 0x2})
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/attacher/bcc_attacher.go:68 +0x97
github.com/sustainable-computing-io/kepler/pkg/attacher.AttachBPFAssets()
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/attacher/bcc_attacher.go:109 +0x265
github.com/sustainable-computing-io/kepler/pkg/collector.(*Collector).Attach(0xc0000a3c10)
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/collector/collector.go:103 +0x33
main.main()
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/cmd/exporter.go:72 +0x40f
```

**Expected behavior**
It would be great if Kepler could:
- handle the error instead of throwing a panic
- support specifying a kernel header version that is higher than the kernel version

**Desktop (please complete the following information):**
 - OS: Ubuntu 2204

rootfs, 2022-09-26:
Missing kernel headers causes eBPF fail to resolve kernel structs. There is a good writing by [Pixie](https://blog.px.dev/ebpf-portability/). We should watch for these new solutions.

jichenjc, 2022-09-27:
just curious and for learning purpose, is it a general question of ebpf so all ebpf related projects has such ?
or the ask is to avoid the panic (so we should do some kind of check ahead) ?

nikimanoledaki, 2022-09-27:
On the one side, there is the ask to avoid the panic in general, which could lead to the other ask, which is to be able to support headers that are a different version than what `uname -r` returns...

For example, in my setup with custom kernel `5.10.77`, I added headers for `5.11.0` with `apt install linux-headers-5.11.0`, which is something that should be supported as well.

See the following from https://wiki.gentoo.org/wiki/Linux-headers:

> A common question regarding linux-headers is whether you need to keep them synced with your kernel version. The answer is no. You can have a newer linux-headers version than your running kernel binary.
> 
> So for example, if you have kernel 4.1 installed, you can have linux-headers 4.4. If you compile your glibc with this newer headers and later upgrade to kernel 4.4, you will be able to use the new features of this kernel without recompiling glibc again.

Really helpful reads on this from the Gentoo wiki: [glibc FAQ](https://sourceware.org/glibc/wiki/FAQ#What_version_of_the_Linux_kernel_headers_should_be_used.3F) and [[2]](https://sourceware.org/ml/libc-help/2014-11/msg00010.html)

You might be able to reproduce this locally by installing a higher kernel header version e.g. `apt install linux-headers-5.11.0`!

rootfs, 2022-09-27:
eBPF is used to get certain stats from kernel for power modeling and inference. The roadmap is to use multiple models, depending on the environment and restriction. On setups that don't have kernel headers, there will be a fallback path to e.g. cgroup/kubelet stats. We can disable the panic once those fallback paths are in place (thanks to @sunya-ch)

nikimanoledaki, 2022-09-27:
Agreed! A note regarding the following, however:

> On setups that don't have kernel headers

If I understand correctly, in my setup kernel headers were installed but they were a different version to the kernel version, which is slightly different from the setup where no kernel headers were installed at all 🤔

rootfs, 2022-09-27:
@nikimanoledaki would you help the documentation with your findings and solutions? That'll bring this issue to a good conclusion

marceloamaral, 2023-05-18:
Can we close this?

nikimanoledaki, 2023-05-18:
The use case where this issue occurred was very specific where I was building the kernel from scratch. This is not something that I work on anymore and may be unlikely to occur again since it is a rare use case. I'd say we can safely close this for now, thank you @marceloamaral!

closed 2023-05-18 by nikimanoledaki