sustainable-computing-io/kepler #252: Attributing I/O energy

rootfs, 2022-09-27:
**Is your feature request related to a problem? Please describe.**
Currently the attribution to Pods is based the Pods CPU/memory activities. The processes in the Pods could have disk I/O activities that are handled by interrupt handlers, we should attribute energy consumed by the handlers properly to the Pods that initiate them.

**Describe the solution you'd like**
The I/O stats are collected by cgroupfs. The interrupt handlers are currently classified as system processes. 
In the first implementation, we can do the following flow
- get the kworker process(es) that handle(s) the disk I/Os, quantify the processes' energy consumption.
- from the process's cgroup disk I/O stat, find the block I/O stats on each block device
- evenly distribute the kworker's energy to all processes based on their I/O stats
- create another metrics for the pods called Pod disk I/O energy

We can quantify other I/O energy in similar way, e.g. network I/O.

rootfs, 2022-09-27:
Per kernel source [here](https://github.com/torvalds/linux/blob/3d7cb6b0/block/blk-core.c#L1260), the block I/O worker threads name regex is `^[kworker/.*kblockd]$`, e.g. `[kworker/93:1H-kblockd]`

marceloamaral, 2022-09-28:
Thanks @rootfs for writing the summary of the idea here!

rootfs, 2022-09-28:
https://github.com/orgs/sustainable-computing-io/teams/kepler-core

commit 956f6c8e (2022-10-31): FIX #252: account kblockd kworker energy separately
commit bd1fc6a1 (2022-10-31): collector: fix #252

account kblockd kworker energy separately
commit bc373453 (2023-01-19): collector: fix #252

account kblockd kworker energy separately
commit e7812d29 (2023-01-20): collector: fix #252

account kblockd kworker energy separately
closed 2023-05-24 by stale[bot]