sustainable-computing-io/kepler #254: CI post action enhance

SamYuan1990, 2022-09-27:
**Is your feature request related to a problem? Please describe.**
- remove post integration test
- image build from base to kepler

**Describe the solution you'd like**
n/A

**Describe alternatives you've considered**
n/A

**Additional context**
n/A

SamYuan1990, 2022-09-27:
@marceloamaral , could you please add some reason in comments for remove post integration test checking?

marceloamaral, 2022-09-28:
I'm not saying that post-integration testing isn't useful, there may be some use cases that will be needed

However, for simplicity's sake for now, I think that our integration test should be run before the merge

SamYuan1990, 2022-09-28:
> I'm not saying that post-integration testing isn't useful, there may be some use cases that will be needed
> 
> However, for simplicity's sake for now, I think that our integration test should be run before the merge

I am going to create a pr for this and let's see.

SamYuan1990, 2022-10-22:
close for now as resolved by #267

closed 2022-10-22 by SamYuan1990