sustainable-computing-io/kepler #258: Selectively export metrics

rootfs, 2022-09-28:
**Is your feature request related to a problem? Please describe.**
Currently all metrics are exported. This will create a long list of metrics when there are many pods.

**Describe the solution you'd like**
The exported metrics can be set during startup as well as runtime, potentially through a configmap

marceloamaral, 2022-09-28:
We have groups of metrics, so that we could enable and disable it via command line parameter

jichenjc, 2022-10-10:
>>> This will create a long list of metrics when there are many pods.

we will switch from pod to container mode soon, I believe this issue will not be impacted by that plan/PR ? 
@marceloamaral

marceloamaral, 2022-10-10:
I don't think we're going to disable power consumption metrics, but there are some extra metrics.

For example, we export hardware counters (HC) of containers. Although we have the logic to disable it if HC is not available, we could also make it optional and disable HC even when it is available.

Also, I think there is a plan to expose some cgroup metrics that we're using, and it will be nice to be able to enable and disable them as well.

jichenjc, 2022-10-11:
ok, thanks, I think at least the counters is eligible for that, thanks for sharing this

jichenjc, 2022-10-14:
>>>I think there is a plan to expose some cgroup metrics that we're using, and it will be nice to be able to enable and disable them as well.

I think we are not there yet, do we?

marceloamaral, 2022-10-14:
Not now....

closed 2023-05-24 by stale[bot]