sustainable-computing-io/kepler #259: Is getActiveCPUs works on VM ?

jichenjc, 2022-09-28:
**Describe the bug**
A clear and concise description of what the bug is.

running on VM env (Ubuntu) 

[getActiveCPUs](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/reader.go#L147)

I printed log like this  after the line and since we are in `attacher.EnableCPUFreq`  else logic, should this have some data anyway?
klog.V(5).Infof("---activecpu---%d----%s", activeCPUs, ct.CPUTime)

and got following
```
I0928 09:37:55.760271       1 reader.go:150] ---activecpu---[]----[%!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0) %!s(uint16=0)]

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

marceloamaral, 2022-09-28:
If the eBPF program is successfully running, the CPU time should be accounted...
Can you check if the eBPF is running?

rootfs, 2022-09-28:
@sustainable-computing-io/kepler-core @sustainable-computing-io/kepler-deployment

jichenjc, 2022-10-08:
>>>Can you check if the eBPF is running?

marceloamaral, 2022-10-11:
> bpf: Failed to load program: Permission denied

Can you give more lines of the log?
It seems that ebpf is not running

jichenjc, 2022-10-14:
I saw some logs like 
```
bpf: Failed to load program: Permission denied
; int sched_switch(switch_args *ctx)
0: (bf) r6 = r1
; u64 pid = bpf_get_current_pid_tgid() >> 32;
1: (85) call bpf_get_current_pid_tgid#14
; u64 pid = bpf_get_current_pid_tgid() >> 32;
2: (77) r0 >>= 32
; u64 pid = bpf_get_current_pid_tgid() >> 32;
3: (7b) *(u64 *)(r10 -8) = r0
; u64 time = bpf_ktime_get_ns();
4: (85) call bpf_ktime_get_ns#5
5: (bf) r7 = r0
; u64 time = bpf_ktime_get_ns();
6: (7b) *(u64 *)(r10 -16) = r7
; u32 cpu_id = bpf_get_smp_processor_id();
7: (85) call bpf_get_smp_processor_id#8
; u32 cpu_id = bpf_get_smp_processor_id();
8: (63) *(u32 *)(r10 -20) = r0
; old_pid.pid = ctx->prev_pid;
9: (61) r1 = *(u32 *)(r6 +24)
; old_pid.pid = ctx->prev_pid;
10: (63) *(u32 *)(r10 -32) = r1
; u64 *last_time = bpf_map_lookup_elem((void *)bpf_pseudo_fd(1, -3), &old_pid);
11: (18) r1 = 0xffff952a5a8b0000
13: (bf) r2 = r10
....
I1014 00:28:55.945051       1 bcc_attacher.go:111] failed to attach perf module with options [-DNUM_CPUS=8 -DCPU_FREQ]: failed to load sched_switch: error loading BPF program: permission denied
```

I see error printed in my log
https://github.com/sustainable-computing-io/kepler/blob/9762279ba65fdddc81d5ec60c8b8adb9dc05ded7/pkg/attacher/bcc_attacher.go#L117
but I didn't se error printed 
https://github.com/sustainable-computing-io/kepler/blob/9762279ba65fdddc81d5ec60c8b8adb9dc05ded7/pkg/attacher/bcc_attacher.go#L122

so I Think eBPF runs correctly as the module loaded? what's the criteria of detecting whether eBPF runs smoothly or not in kepler context? Thanks

marceloamaral, 2022-10-14:
Right, the logs aren't saying much....We need to improve it.
We should have a log saying in the end if the bpf is running or not....

But regarding the first failure. 
Actually the `-DCPU_FREQ` parameter is related to CPU time, which is used to calculate the frequency later...

So the first attempt to attach the bpf program is with the support to collect cpu time....
If it fails for some reason, it will run a program without the CPU time....

Btw, can you [pastebin](https://pastebin.com/) the full log?

In fact, I want to deprecate the CPU time since the cpu time is also reported by cgroup...
And then, I will update the logic to calculate the frequency later.... This is in my TODO list...

marceloamaral, 2023-05-18:
Closing it because we don't have this problem anymore.

closed 2023-05-18 by marceloamaral