sustainable-computing-io/kepler #261: Separating system processes from Pod energy metrics

rootfs, 2022-09-28:
**Is your feature request related to a problem? Please describe.**
Currently for consistency, all the system processes are classified as system processes Pod and in system namespace. 
This is a problem for end user and becomes unscalable when further separating kblockd and softirq from the system processes.

**Describe the solution you'd like**
The system processes, kblockd, ksoftirqd and any upcoming non-Pod identities should have their own metrics, not to be mixed in the Pods metrics.

rootfs, 2022-09-28:
> We can have this information in the same metric, as it will be easier to aggregate and manipulate the information in Prometheus But instead of using the pod name label, we should leave that label empty and have another label called process_name Wdyt?

sounds good!

jichenjc, 2022-10-10:
I saw there are multiple PR/issues to separate the kblockd etc, so any todo for this issue (only) ?

marceloamaral, 2022-10-11:
This is a WIP and needs more design thinking

closed 2023-05-24 by stale[bot]