sustainable-computing-io/kepler #262: Grafana dashboard datasource 'prometheus' not found

Feelas, 2022-09-28:
**Describe the bug**
Creating Kepler dashboard using OpenShift scripts succeeds but using it fails since it cannot find 'prometheus' datasource. Manually changing datasource name from "prometheus" to "Prometheus" and then importing fixes the problem. This is probably after changes from #201. It would be good to have a working example, which doesn't require the user to modify the dashboard by hand to deploy.

**To Reproduce**
After installing Kepler on OpenShift, run manifests/openshift/dashboard/deploy-grafana.sh

**Expected behavior**
The dashboard is created successfully. We can either revert the #201 (read additional context) or change the hardcoded name in openshift manifest scripts to lowercase 'prometheus'. 

**Additional context**
Problem happens because of manifests/openshift/03-grafana-datasource-UPDATETHIS.yaml defines "Prometheus", not "prometheus". Grafana dashboard name matching is case sensitive (https://github.com/grafana/grafana/issues/23067).

The previous behaviour with "datasource": null is supposed to use the default specified datasource (https://github.com/grafana/grafana/issues/19190), it would make sense to rollback the change and instruct the users on how to properly mark a data source as default instead of hard-coding a datasource name.

cc @kenplusplus, can you try marking your datasource as default and then using "datasource": null dashboard, as was previously defined?

rootfs, 2022-10-04:
@sallyom this is probably the issue seen on your setup

closed 2022-10-04 by rootfs

jeremycohn, 2022-10-05:
Can confirm this fixed issue in OpenShift.