sustainable-computing-io/kepler #264: no collection of new deployed pod on VM based env?

jichenjc, 2022-09-29:
**Describe the bug**
A clear and concise description of what the bug is.

I had following k8s , the nginx n1/n2 namespace pods are deployed after kepler daemonset start running 

```
# kubectl get pods -A
NAMESPACE     NAME                                       READY   STATUS    RESTARTS      AGE
....
kepler        kepler-exporter-66zfj                      1/1     Running   0             47m
kube-system   calico-kube-controllers-54c85446d4-pb549   1/1     Running   2 (46h ago)   8d
....
logging       fluentd-es-hlt4c                           1/1     Running   1 (46h ago)   2d12h
logging       kibana-568dfb5bf-bgrt9                     1/1     Running   1 (46h ago)   2d13h
monitoring    alertmanager-main-0                        2/2     Running   2 (46h ago)   8d
monitoring    alertmanager-main-1                        2/2     Running   2 (46h ago)   8d
monitoring    alertmanager-main-2                        2/2     Running   2 (46h ago)   8d
monitoring    blackbox-exporter-6d897d5f66-dtspw         3/3     Running   3 (46h ago)   8d
monitoring    grafana-59664d747b-tj274                   1/1     Running   1 (46h ago)   6d17h
monitoring    kube-state-metrics-776dd9bc5d-rx4bq        3/3     Running   3 (46h ago)   8d
monitoring    node-exporter-thrwp                        2/2     Running   2 (46h ago)   8d
monitoring    prometheus-adapter-867598997b-dphc9        1/1     Running   1 (46h ago)   8d
monitoring    prometheus-adapter-867598997b-kstng        1/1     Running   1 (46h ago)   8d
monitoring    prometheus-k8s-0                           2/2     Running   2 (46h ago)   8d
monitoring    prometheus-k8s-1                           2/2     Running   2 (46h ago)   8d
monitoring    prometheus-operator-65cdf86899-bd4rl       2/2     Running   2 (46h ago)   8d
n1            nginx-deployment-7fb96c846b-8tpmd          1/1     Running   0             13m
n1            nginx-deployment-7fb96c846b-g67pz          1/1     Running   0             13m
n2            nginx-deployment-7fb96c846b-74ctr          1/1     Running   0             13m
n2            nginx-deployment-7fb96c846b-q7hsl          1/1     Running   0             13m

```

note n1 and n2 are nginx I want to do other test.. but I can't see nginx repoted at all

```
# curl localhost:9102/metrics |grep millijoule  | grep nginx
...
#
```

I even killed the pod and restart it 

```
kepler        kepler-exporter-lxwmh                      1/1     Running   0             4m14s

```

but I still get same result

I can see etcd 

```
root@jitest40:~# curl localhost:9102/metrics |grep millijoule  | grep etcd
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 75532    0 75532    0     0  1pod_curr_dynamic_energy_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 0
8.0Mpod_curr_energy_in_core_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 1
      0 --:--:-- --:--:-- --pod_curr_energy_in_dram_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 28
:--:pod_curr_energy_in_gpu_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 0
-- 1pod_curr_energy_in_other_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 0
8.pod_curr_energy_in_pkg_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 1
0M
pod_curr_energy_in_uncore_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 0
pod_curr_energy_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 1
pod_total_dynamic_energy_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 0
pod_total_energy_in_core_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 105
pod_total_energy_in_dram_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 2581
pod_total_energy_in_gpu_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 0
pod_total_energy_in_other_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 0
pod_total_energy_in_pkg_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 105
pod_total_energy_in_uncore_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 0
pod_total_energy_millijoule{command="etcd",pod_name="etcd-jitest40",pod_namespace="kube-system"} 105

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

rootfs, 2022-09-29:
@sunya-ch can you take a look? I think the stats are reset for some reason, it doesn't show all the time.

marceloamaral, 2022-09-29:
Cloud you please tell more details of the environment?
Are you using the kind cluster?
BM -> VM -> Kind?

jichenjc, 2022-09-29:
no, VM only , no kind ...

```
# export KUBECONFIG=/etc/kubernetes/admin.conf
root@jitest40:~/m# cat /etc/kubernetes/admin.conf | grep 10.43
    server: https://10.43.0.203:6443
root@jitest40:~/m# ip a | grep 10.43
    inet 10.43.0.203/24 brd 10.43.0.255 scope global dynamic ens3

```

rootfs, 2022-09-29:
this happens to my setup too, baremetal. Even nginx finally showed up, the command was wrong
```console
[root@clever src]# curl localhost:9102/metrics 2>&1 | grep nginx-798cf574f5-r6d5g
[root@clever src]# curl localhost:9102/metrics 2>&1 | grep nginx-798cf574f5-r6d5g
[root@clever src]# curl localhost:9102/metrics 2>&1 | grep nginx-798cf574f5-r6d5g
[root@clever src]# curl localhost:9102/metrics 2>&1 | grep nginx-798cf574f5-r6d5g
[root@clever src]# curl localhost:9102/metrics 2>&1 | grep nginx-798cf574f5-r6d5g
[root@clever src]# curl localhost:9102/metrics 2>&1 | grep nginx-798cf574f5-r6d5g
pod_cpu_instructions{pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 644670
pod_curr_dynamic_energy_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 0
pod_curr_energy_in_core_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 0
pod_curr_energy_in_dram_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 1
pod_curr_energy_in_gpu_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 0
pod_curr_energy_in_other_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 0
pod_curr_energy_in_pkg_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 51
pod_curr_energy_in_uncore_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 0
pod_curr_energy_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 51
pod_energy_stat{avg_cpu_frequency="1925920.645833",block_devices_used="1",command="conmon",curr_bytes_read="0",curr_bytes_writes="0",curr_cache_miss="2133",curr_container_cpu_usage_seconds_total="0",curr_container_memory_working_set_bytes="0",curr_cpu_cycles="3238378",curr_cpu_instr="644670",curr_cpu_time="0",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1",total_bytes_read="0",total_bytes_writes="19177472",total_cache_miss="2133",total_container_cpu_usage_seconds_total="0",total_container_memory_working_set_bytes="70713344",total_cpu_cycles="3238378",total_cpu_instr="644670",total_cpu_time="0"} 51
pod_total_dynamic_energy_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 0
pod_total_energy_in_core_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 0
pod_total_energy_in_dram_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 1
pod_total_energy_in_gpu_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 0
pod_total_energy_in_other_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 0
pod_total_energy_in_pkg_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 51
pod_total_energy_in_uncore_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 0
pod_total_energy_millijoule{command="conmon",pod_name="nginx-798cf574f5-r6d5g",pod_namespace="test-1"} 51
```

rootfs, 2022-09-29:
there could be two possible causes:
1) the pid/cgroup -> pod resolution is wrong (so the wrong conmon was picked up)
2) the stats were reset (the `pod_total_energy_millijoule` should stay on regardless of current energy)

sunya-ch, 2022-09-29:
> there could be two possible causes:
> 
> 1. the pid/cgroup -> pod resolution is wrong (so the wrong command was picked up)
> 2. the stats were reset (the `pod_total_energy_millijoule` should stay on regardless of current energy)

I think the podEnergy is reset as there is no context_switch of that pod detected by BPF here https://github.com/sustainable-computing-io/kepler/blob/fc425e939332b659a376c933160afe52d1fb03ad/pkg/collector/reader.go#L201-L205.

It is in the to-fix list that we should double-check whether the pod is deleted or just no activity. 
For the command shown up, it is the latest command called by that pod which might be not the main one.

https://github.com/sustainable-computing-io/kepler/blob/fc425e939332b659a376c933160afe52d1fb03ad/pkg/collector/reader.go#L137

We may instead get the other meaningful information from the kubelet instead of the command from BPF.

marceloamaral, 2022-09-29:
I had a similar issue because the pod information was being deleted and then I commented out the part that deletes the pod information. during my experiments... However, the memory usage may grow and never be freed even if the pod has been deleted....

So instead of removing pod information from `podEnergy` when there is no HC update, we could introduce a timeout
For example, if we don't see any updates on the pod's usage for at least 5 minutes, we can assume the pod is no longer active and delete it from `podEnergy`

rootfs, 2022-09-29:
@marceloamaral how about introduce a garbage collector to remove metrics from inactive pods when the metrics map size (to be configured) is too large?

jichenjc, 2022-10-08:
I assume this has been tested already with the PR ?if so we can close this

marceloamaral, 2022-10-08:
@jichenjc can you test if the issue was solved?

jichenjc, 2022-10-08:
ok, then it's not newly created pod, it's inactive pod .. 
anyway ,seems deploy a new deployment with nginx pods I can see them from output,so close this

closed 2022-10-08 by jichenjc