sustainable-computing-io/kepler #274: remove vendor folder and modules.txt, or latest remove github.com/onsi/ginkgo from it

SamYuan1990, 2022-10-02:
**Is your feature request related to a problem? Please describe.**
I would like to remove remove vendor folder and modules.txt, or latest remove github.com/onsi/ginkgo from it.

**Describe the solution you'd like**
You know, there no need to add github.com/onsi/ginkgo in vendor folder for fix it. And why we need vendor folder?

**Describe alternatives you've considered**
n/A

**Additional context**
n/A

SamYuan1990, 2022-10-02:
btw, we should update ginkgo to v2.

sunya-ch, 2022-10-03:
For the reason to keep the vendor folder, @rootfs already mentioned about it here.
> I prefer to keep the vendor directory, in case the vendor pkgs are missing and it saves time on CI.
https://github.com/sustainable-computing-io/kepler/pull/134#issuecomment-1238050399

As long as `make test` is working, I think any case would be fine. 
The point is to make the ginkgo binary available for running the test.

marceloamaral, 2022-10-03:
Vendor folder has only 28M, I think it is ok to keep it

SamYuan1990, 2022-10-03:
> For the reason to keep the vendor folder, @rootfs already mentioned about it here.
> 
> > I prefer to keep the vendor directory, in case the vendor pkgs are missing and it saves time on CI.
> > [#134 (comment)](https://github.com/sustainable-computing-io/kepler/pull/134#issuecomment-1238050399)
> 
> As long as `make test` is working, I think any case would be fine. The point is to make the ginkgo binary available for running the test.

@marceloamaral , @sunya-ch , @rootfs I can understand that we hope to keep some files 
> in case the vendor pkgs are missing and it saves time on CI.
but, for 
`github.com/onsi/ginkgo` and `github.com/onsi/gomega`
1. they are just use as test framework.
2. so far ginkgo is upgrade to v2.
3. we missing `github.com/onsi/gomega/gexec` at https://github.com/sustainable-computing-io/kepler/blob/main/vendor/modules.txt#L85-L96, as it doesn't have all packages.
> the reason why I wanna to have  `github.com/onsi/gomega/gexec`, is that I wanna to have an end to end test case run at
> - build kepler locally and run with `os.exe` 
> - by environment value, switch test case targeting on k8s cluster, once after kepler been deployed in CI.

Hence I want to remove vendor folder, or at least remove ginkgo and gomega from vendor.

marceloamaral, 2022-10-03:
I thought that the following commands would fix those problems....
```
go mod tidy
go mod vendor
```

rootfs, 2022-10-03:
run it on the current source tree, it is clean after tidy
```console
]# go mod tidy
# go mod vendor
# git status
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```

SamYuan1990, 2022-10-03:
k, close this one for now.
will use command below instead.
```
go mod tidy
go mod vendor
```