sustainable-computing-io/kepler #276: enhance reader.go logic with *enableGPU parameter

SamYuan1990, 2022-10-03:
so far if there no gpu device, the code still go through with gpu logic for example: `podGPUDelta`
as we have `*enableGPU`, if `GPU` is disabled, can we skip gpu related logic?

_Originally posted by @SamYuan1990 in https://github.com/sustainable-computing-io/kepler/discussions/273#discussioncomment-3787199_

marceloamaral, 2022-10-04:
I think the logic should try to discover gpus.
So, if there is no gpus in the system we disable the calculation.

Will you submit a PR for this?

SamYuan1990, 2022-10-04:
> I think the logic should try to discover gpus. So, if there is no gpus in the system we disable the calculation.
> 
> Will you submit a PR for this?

not for now @marceloamaral , I am no local holiday...
if you want it, you can pick it up for now.

rootfs, 2022-10-04:
@SamYuan1990 enjoy the holiday!

marceloamaral, 2022-10-08:
The PR #287 fix this

closed 2022-10-24 by SamYuan1990