sustainable-computing-io/kepler #282: make bcc logic part more robust

SamYuan1990, 2022-10-07:
we have an issue as we installed bcc at container, but if host vm/instance don't bcc lib installed, and bcc been used as volume mount of pod, the pod will exited as bcc not found.

>
        @rootfs , @marceloamaral , take a look at https://github.com/sustainable-computing-io/kepler/actions/runs/3201921603/jobs/5230372207#step:8:13 pls

_Originally posted by @SamYuan1990 in https://github.com/sustainable-computing-io/kepler/issues/275#issuecomment-1271050426_

SamYuan1990, 2022-10-07:
```
Run kubectl logs $(kubectl -n kepler get pods -oname) -n kepler
I1007 03:00:18.[4](https://github.com/sustainable-computing-io/kepler/actions/runs/3201921603/jobs/5230372207#step:8:5)9110[5](https://github.com/sustainable-computing-io/kepler/actions/runs/3201921603/jobs/5230372207#step:8:6)       1 config.go:62] config EnabledEBPFCgroupID enabled:  true
I1007 03:00:18.491172       1 config.go:63] config getKernelVersion:  5.15
I1007 03:00:18.491193       1 config.go:[6](https://github.com/sustainable-computing-io/kepler/actions/runs/3201921603/jobs/5230372207#step:8:7)[7](https://github.com/sustainable-computing-io/kepler/actions/runs/3201921603/jobs/5230372207#step:8:8)] config set EnabledEBPFCgroupID to  false
modprobe: FATAL: Module kheaders not found in directory /lib/modules/5.15.0-1020-azure
chdir(/lib/modules/5.15.0-1020-azure/build): No such file or directory
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x[8](https://github.com/sustainable-computing-io/kepler/actions/runs/3201921603/jobs/5230372207#step:8:9) pc=0x81ae56]

goroutine 1 [running]:
github.com/iovisor/gobpf/bcc.(*Module).Load(0x0, {0xd372f4, 0xc}, 0x2?, 0x2?, 0xc000434d60?)
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/vendor/github.com/iovisor/gobpf/bcc/module.go:202 +0x36
github.com/iovisor/gobpf/bcc.(*Module).LoadTracepoint(...)
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/vendor/github.com/iovisor/gobpf/bcc/module.go:182
github.com/sustainable-computing-io/kepler/pkg/attacher.loadModule({0x15c1420?, 0xd35e7f?, 0x0?}, {0xc000073800, 0x2, 0x2})
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/attacher/bcc_attacher.go:68 +0x[9](https://github.com/sustainable-computing-io/kepler/actions/runs/3201921603/jobs/5230372207#step:8:10)7
github.com/sustainable-computing-io/kepler/pkg/attacher.AttachBPFAssets()
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/attacher/bcc_attacher.go:[10](https://github.com/sustainable-computing-io/kepler/actions/runs/3201921603/jobs/5230372207#step:8:11)9 +0x265
github.com/sustainable-computing-io/kepler/pkg/collector.(*Collector).Attach(0xc000424c90)
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/collector/collector.go:104 +0x33
main.main()
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/cmd/exporter.go:80 +0x40f
```

rootfs, 2022-10-07:
`Module kheaders not found in directory /lib/modules/5.15.0-1020-azure`
which kernel did you use? @SamYuan1990

SamYuan1990, 2022-10-07:
> `Module kheaders not found in directory /lib/modules/5.15.0-1020-azure` which kernel did you use? @SamYuan1990

not for now, I didn't take further investigate.

marceloamaral, 2022-10-08:
@rootfs this is from the Github VM, which seems to be from azure....

closed 2023-05-24 by stale[bot]