sustainable-computing-io/kepler #286: Prometheus metrics are not conforming with the Prometheus metrics guideline

marceloamaral, 2022-10-07:
**Is your feature request related to a problem? Please describe.**
Currently the prometheus metrics are not conforming with the [Prometheus metrics guideline](https://prometheus.io/docs/practices/naming/), and more details [here](https://prometheus.io/docs/instrumenting/writing_exporters/).

The key rules that we are violating are:
- should use base units (e.g. seconds, bytes, meters - not milliseconds, megabytes, kilometers)
- should have a suffix describing the unit, in plural form
- `_total` is a convention for counters, you should use it if you’re using the COUNTER type.
- Prefer exporting a counter of joules, then rate(joules[5m]) gives you power in Watts.
- metric names should allow someone who is familiar with Prometheus but not a particular system to make a good guess as to what a metric means.
- metric names should be avoided to be procedurally generated
- metric names for applications should generally be prefixed by the exporter name, e.g. `kepler_`
- metric should make sense when summed or averaged, e.g. we should not have package energy with core energy
- use labels to differentiate the characteristics, that is, they should not contain values, e.g., cpu instructions

**Describe the solution you'd like**
Replace the old metric names with new metric ones.
For the sake of compatibility with other modules, we should also keep some deprecated metrics and remove it later.
The new metrics should be:
```
//  new metrics
kepler_node_info{cpu_architecture}
kepler_node_core_joules_total{package, instance, source="rapl"}
kepler_node_uncore_joules_total{package, instance, source="rapl"}
kepler_node_dram_joules_total{package, instance, source="rapl"}
kepler_node_package_joules_total{package, instance, source="rapl"}
kepler_node_platform_joules_total{instance="node1", source="hwmon"}
kepler_node_other_host_components_joules_total{instance="node1"}
kepler_node_gpu_joules_total{index, instance, source="nvidia"}
kepler_node_cpu_scaling_frequency_hertz{instance, core}

kepler_pod_core_joules_total{pod_name="", pod_namespace="", command=""}
kepler_pod_uncore_joules_total{pod_name="", pod_namespace="", command=""}
kepler_pod_dram_joules_total{pod_name="", pod_namespace="", command=""}
kepler_pod_package_joules_total{pod_name="", pod_namespace="", command=""}
kepler_pod_other_host_components_joules_total{pod_name="", pod_namespace="", command=""}
kepler_pod_gpu_joules_total{pod_name="", pod_namespace="", command=""}
kepler_pod_cpu_cycles_total{pod_name="", pod_namespace="", command=""}
kepler_pod_cpu_instr_total{pod_name="", pod_namespace="", command=""}
kepler_pod_cache_miss_total{pod_name="", pod_namespace="", command=""}
```

```
// deprecated pod metrics
pod_cpu_cpu_time_us{pod_name, pod_namespace, cpu}
kepler_pod_energy_stat{block_devices_used, command, curr_bytes_read, curr_bytes_writes, curr_cache_miss, curr_container_cpu_usage_seconds_total, curr_container_memory_working_set_bytes, curr_cpu_cycles, curr_cpu_instr, curr_cpu_time, pod_name, pod_namespace, total_bytes_read, total_bytes_writes, total_cache_miss, total_container_cpu_usage_seconds_total, total_container_memory_working_set_bytes, total_cpu_cycles, total_cpu_instr, total_cpu_time}

// deprecated node metrics
node_package_energy_millijoule{instance, pkg_id, core, dram, uncore}
node_energy_stat{node_name, cpu_architecture, node_curr_cpu_time, node_curr_cpu_cycles, node_curr_cpu_instr, node_curr_cache_miss, node_curr_container_cpu_usage_seconds_total, node_curr_container_memory_working_set_bytes, node_curr_bytes_read, node_curr_bytes_writes, node_block_devices_used, node_curr_energy_in_core_joule, node_curr_energy_in_dram_joule, node_curr_energy_in_gpu_joule, node_curr_energy_in_other_joule, node_curr_energy_in_pkg_joule, node_curr_energy_in_uncore_joule}
```

**Describe alternatives you've considered**
Regarding the `Prefer exporting a counter of joules, then rate(joules[5m]) gives you power in Watts.`
Initially, I was considering that it would be better to export both watts and joules, specially because Kepler aggregates values of a 3 s interval.

So, I have tested two different metrics.
The metric `kepler_pod_package_watts{}` is in the first figure below
- this metric export the pod last measured energy consumption already divided by the kepler sample interval  

The metric `rate(kepler_pod_package_joules_total{}[4s])` is in the second figure below
- this metric reports the aggregated energy consumption
- the prometheus rate function returns the values per second
- the interval [4s] will force show all points, a larger interval will just make the curve more smooth

We can see that both functions give very approximated values, so that we don't need to export watts metrics, only joules.
Please, also note that we don't need to divide the values of the second function by the kepler sampling interval because this function already gives the value per second.

[image]

**Additional context**
I have also review the metric naming in the [node-exporter project](https://github.com/prometheus/node_exporter/blob/master/collector/fixtures/e2e-64k-page-output.txt).

marceloamaral, 2022-10-08:
Also, note that the goal is to update the metrics again later, we should export container energy consumption and not pod. Like `container_cpu_usage_seconds_total`, but we will have `kepler_container_package_joules_total`...
Did not use the container name right now because it will require much more changes in the code...

marceloamaral, 2022-10-08:
So we will have the energy consumption per container as well as. In the figure below the labels are `{{pod_name}} - {{container_name}} - {{resource}}`
[image]

rootfs, 2022-10-13:
@marceloamaral Let's keep the conformance and usability goals in sync. The new metrics are good for query and dashboarding. The `pod_cpu_cpu_time_us`, `node_package_energy_millijoule`, and `node_energy_stat` are used by the model server and autoscalers to build models. In that use case, query to customized metric gets all the related stats. 

How about we distinguish these use cases and expose the metrics in this way? The proposed metrics format are documented and recommended for general dashboard and query. The compound, customized metrics are used for modeling and other special purposes. These metrics are prefixed so not to be meant for end users.

rootfs, 2022-10-13:
@sustainable-computing-io/reviewer

williamcaban, 2022-10-13:
With the alignment to Prometheus metrics guidelines I would add the ask with the refactoring to include with all metrics the label `pod_node=""` as the energy consumption of a workload is impacted by the particularities of the node where it is running (e.g. node load, temperature, CPU family or version, and others). Such label will also help facilitating better queries of a particular metric for all objects associated with a node.

marceloamaral, 2022-10-14:
Thanks @rootfs for the review!

> The compound, customized metrics are used for modeling and other special purposes. These metrics are prefixed so not to be meant for end users.

I think the model server should be updated to collect metrics using different prometheus queries.
For example, compound metrics have hardware counter and cgroup metrics. And all these metrics will be accessible through their own Prometheus metric. In the end this will be more transparent and easier to understand the model server....
Then we can deprecated these metrics. This will also help decrease the number of metrics we expose

> We should also look at metrics from a scalability perspective.

I agree and this is a typical challenge for prometheus exporters :)
So the option to disable some metrics (eg hc) when needed can help with that.

Also, most of the metrics from the cgroup are already exported to prometheus, so we probably don't need to export them and the model server will get these metrics from the classic metrics.
But note that, the classic metrics may have some difference in value because they are not synced with the kepler collection... This is something we need to investigate....

marceloamaral, 2022-10-14:
Thank you @williamcaban

>include in all metrics the label pod_node=""

I think that makes a lot of sense!
Most metrics that export the node have `instance` and/or `node` labels.
Is it ok if we use `instance`?

marceloamaral, 2022-10-14:
Btw, I just saw that the [clever](https://github.com/sustainable-computing-io/clever) dashboard use [pod_cpu_instructions](https://github.com/sustainable-computing-io/clever/blob/main/dashboards/clever-dashboard-w-IPS.json#L329).

Since we are going to export `container_cpu_instructions` we will need to update the clever dashboard as well as.
Because this is very simple to updater, I don't think we need to keep the old metric...

rootfs, 2022-10-14:
> Also, most of the metrics from the cgroup are already exported to prometheus, so we probably don't need to export them and the model server will get these metrics from the classic metrics.
But note that, the classic metrics may have some difference in exported interval because they are not synced with the kepler collection... This is something we need to investigate....

On deployment without other exporters (due to resource reason), the cgroup or node metrics will be missing. 

I think we all agree the options to turn on/off metrics are important. It will be the way to manage conformance, scalability, and potential overlapping with other exporters. Can we have that option first and we add/enable/disable metrics based on the option.

jichenjc, 2022-10-17:
>>>could be as high as 200. That means kepler exposes 20,000 metrics per node. It will create lots of them in a large cluster

forgive my misunderstanding, is it 2000 instead of 20000?

marceloamaral, 2023-05-18:
We fixed this.

closed 2023-05-18 by marceloamaral