sustainable-computing-io/kepler #289: is CPU instr etc metrics available on VM?

jichenjc, 2022-10-10:
**Describe the bug**
A clear and concise description of what the bug is.

might be related to https://github.com/sustainable-computing-io/kepler/issues/259

in VM (VM running on BM, no kind) 

All logs indicate no cpu actions 
```
counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instr:0 (0)]
```

1) why all instr are 0 (seems from cgroup) ,is it expected?
2) and curious on why we only report pod_cpu_instructions but no pod_cpu_cycles etc into metrics?
```
pod_cpu_instructions{pod_name="alertmanager-main-0",pod_namespace="monitoring"} 0
pod_cpu_instructions{pod_name="alertmanager-main-1",pod_namespace="monitoring"} 0
pod_cpu_instructions{pod_name="alertmanager-main-2",pod_namespace="monitoring"} 0
pod_cpu_instructions{pod_name="blackbox-exporter-6d897d5f66-dtspw",pod_namespace="monitoring"} 0
pod_cpu_instructions{pod_name="calico-kube-controllers-54c85446d4-pb549",pod_namespace="kube-system"} 0
pod_cpu_instructions{pod_name="calico-node-4klcf",pod_namespace="kube-system"} 0
pod_cpu_instructions{pod_name="coredns-565d847f94-cpg9l",pod_namespace="kube-system"} 0
pod_cpu_instructions{pod_name="coredns-565d847f94-gx5l6",pod_namespace="kube-system"} 0
pod_cpu_instructions{pod_name="counter",pod_namespace="default"} 0
pod_cpu_instructions{pod_name="counter",pod_namespace="n1"} 0
pod_cpu_instructions{pod_name="es-0",pod_namespace="logging"} 0
pod_cpu_instructions{pod_name="es-1",pod_namespace="logging"} 0
pod_cpu_instructions{pod_name="es-2",pod_namespace="logging"} 0
pod_cpu_instructions{pod_name="etcd-jitest40",pod_namespace="kube-system"} 0
pod_cpu_instructions{pod_name="fluentd-es-hlt4c",pod_namespace="logging"} 0

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

rootfs, 2022-10-13:
@jichenjc `pod_cpu_instructions` is used for VPA dashboard [here](https://github.com/sustainable-computing-io/clever/blob/main/dashboards/clever-dashboard-w-IPS.json#L329)

jichenjc, 2022-10-14:
>>>   The VM expose hardware counters. In theory, we just need to enable cpu passthrough in the VM. But it is something in my TODO list to verify.... Btw, VMs on amazon exposes hc...
    The bpf program is running. We had some previous PRs that showed that a VM did not have the kernel libs, but to collect hc, we need to make sure that the bpf program can run smoothly...

ok, this seems to be a follow up action for VM
as lack of BM env, I have to cont with VM for a while :( 
but good to know we have a gap here

marceloamaral, 2022-12-15:
@jichenjc can we close this?

jichenjc, 2022-12-15:
yes, let's close

closed 2022-12-15 by jichenjc