sustainable-computing-io/kepler #298: build release version container images and update deployment manifests

rootfs, 2022-10-13:
**Is your feature request related to a problem? Please describe.**
Current the kepler container image is built on main branch and the tag is latest. The tag is used in kepler deployment manifests

**Describe the solution you'd like**
In release branch (e.g. branch name is `release-x.x`, the CI should build the image and use the release tag, updates the manifests once the image is pushed.

**Describe alternatives you've considered**
This can be done manually but CI is better suited for this purpose.

**Additional context**
Add any other context or screenshots about the feature request here.

rootfs, 2022-10-13:
we can use bot to update the manifest once a release image is built

jichenjc, 2022-10-14:
I think this can also be integrated with github actions @SamYuan1990  @rootfs 
e.g every time a new release come out we cut the branch and let github actions tag and build image then upload

this might be FYI as we did it every k8s release 
https://github.com/kubernetes/cloud-provider-openstack/blob/master/release-procedure.md

closed 2023-05-24 by stale[bot]