sustainable-computing-io/kepler #299: Add flag to enable VM deployment during CI

jichenjc, 2022-10-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

https://github.com/sustainable-computing-io/kepler/pull/294#issuecomment-1278354596

I am very curious why the current deployment can pass ,though @marceloamaral @rootfs 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

marceloamaral, 2022-10-14:
The PR #294 does not deploy Kepler to nodes that have the label:
`feature.node.kubernetes.io/cpu-cpuid.HYPERVISOR`

So, as I mentioned, this is very specific to someone's infrastructure....
This means that other infrastructures, even if they are running on top of VMs, may not have this label...

rootfs, 2022-10-14:
cc @ruomengh

jichenjc, 2022-10-16:
Added a couple of comments in #294 
in summary, I think we should agree there before we do any actions here

ruomengh, 2022-10-17:
Actually this depends on NFD (node feature discovery) and this NFD works for x86 architecture only. Please refer to https://kubernetes-sigs.github.io/node-feature-discovery/v0.11/get-started/features.html#arm-cpuid-flags-partial-list
This could affect environments which don't have NFD deployed or which are not x86.

jichenjc, 2022-10-17:
yes, I found that and commented in #294 with exactly same thing you described

I am not sure whether we need introduce such dependency and deployment for such minor feature 
happy to get folks to comment...

marceloamaral, 2023-05-18:
I am closing this issue since we agreed to not add such a flag.
But we can open it again if needed.

closed 2023-05-18 by marceloamaral