sustainable-computing-io/kepler #301: aggregate energy consumption from Pod level to higher level

rootfs, 2022-10-16:
**Is your feature request related to a problem? Please describe.**
Pod level energy reporting is useful on a per node basis. At the cluster level, high level aggregation is more useful.

**Describe the solution you'd like**
When calling [`ListPods`](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/podlister/resolve_container.go#L131),  `OwnerReferences` in the Pod metadata should be parsed to get the Pods' owner. 

**Describe alternatives you've considered**

**Additional context**
Add any other context or screenshots about the feature request here.

jichenjc, 2022-10-17:
maybe `per application (OCP concept) or deployment level` as user might pay attention to the workload 
and daemonset consumption per cluster is a good measure to side effect from power perspective?

marceloamaral, 2022-10-17:
This is very useful, specially to report `Job` energy consumption

rootfs, 2022-10-17:
For Deployment Pods, there will be steps I think:
- at kubelet level, we can get the Pod owner. This works for certain API objects such as DaemonSet/ReplicaSet, not to Deployment,etc.  This query should be isolated to individual nodes, will not stress the API server.
- If we want to go the eventual owner of the ReplicaSet, we need to query API server and go through the ReplicaSet's OwnerReference to find the Deployment info. This query goes to the API server but since at the ReplicaSet level, the traffic is less frequent.

jichenjc, 2022-10-18:
>>>at kubelet level, we can get the Pod owner. This works for certain API objects such as DaemonSet/ReplicaSet, not to Deployment,etc. This query should be isolated to individual nodes, will not stress the API server.
    If we want to go the eventual owner of the ReplicaSet, we need to query API server and go through the ReplicaSet's OwnerReference to find the Deployment info. This query goes to the API server but since at the ReplicaSet level, the traffic is less frequent.

ok, this seems reasonable, maybe we can go from Daemonset first as it should be simpler

rootfs, 2022-10-18:
@jichenjc sounds good, PR welcome

marceloamaral, 2022-10-18:
Performing get operations for the API server can be expensive on a large scale....

How about creating a watcher with the node label as an event filter....
This should have very low overhead....

So we do the top-down approach instead of the bottom-up approach.

----
I like that this direction with watcher as this will also later support the challenge of identifying if a resource has been deleted

----
An example to watch pods, but the idea is to watch the other resources
```golang

import (
        k8sv1 "k8s.io/api/core/v1"
	"k8s.io/client-go/tools/cache"
        "k8s.io/apimachinery/pkg/fields"
)

// NewListWatchFromClient creates a new ListWatch from the specified client, resource, kubevirtNamespace and field selector.
func NewListWatchFromClient(c cache.Getter, resource string, namespace string, fieldSelector fields.Selector, labelSelector labels.Selector) *cache.ListWatch {
	listFunc := func(options metav1.ListOptions) (runtime.Object, error) {
		options.FieldSelector = fieldSelector.String()
		options.LabelSelector = labelSelector.String()
		return c.Get().
			Namespace(namespace).
			Resource(resource).
			VersionedParams(&options, metav1.ParameterCodec).
			Do(context.Background()).
			Get()
	}
	watchFunc := func(options metav1.ListOptions) (watch.Interface, error) {
		options.FieldSelector = fieldSelector.String()
		options.LabelSelector = labelSelector.String()
		options.Watch = true
		return c.Get().
			Namespace(namespace).
			Resource(resource).
			VersionedParams(&options, metav1.ParameterCodec).
			Watch(context.Background())
	}
	return &cache.ListWatch{ListFunc: listFunc, WatchFunc: watchFunc}
}

func (w *ObjListWatcher) Run() {
	go w.informer.Run(w.stopChannel)
	timeoutCh := make(chan struct{})
	timeoutTimer := time.AfterFunc(informerTimeout, func() {
		close(timeoutCh)
	})
	defer timeoutTimer.Stop()
	if !cache.WaitForCacheSync(timeoutCh, w.informer.HasSynced) {
		klog.Fatalf("watcher timed out waiting for caches to sync"))
	}
}

listOpts := &metav1.ListOptions{}
listOpts.LabelSelector = fmt.Sprintf("kubernetes.io/hostname=%s", nodename) // to filter events per node
labelSelector, err := labels.Parse(listOpts.LabelSelector)
objListWatcher := NewListWatchFromClient(
		k8sCli.RestClient(),
		"Pod",
		k8sv1.NamespaceAll,
		fields.Everything(),
		labelSelector)

informer = cache.NewSharedInformer(objListWatcher, nil, 0)

informer.AddEventHandler(cache.ResourceEventHandlerFuncs{
		AddFunc: func(oldObj, newObj interface{}) {
			handleAdd(newObj) // update our list
		},
		DeleteFunc: func(obj interface{}) {
			handleDeleted(obj)  // update our list
		},
	})

objWatcher = NewObjListWatcher(informer)
objWatcher.Run()
```

rootfs, 2022-10-18:
@marceloamaral this is a good idea! 

Along the scalability line, having a node watcher on each node requires additional resources and privilege. I am thinking creating yet another exporter (optionally running as yet another deployment) using your pod watcher idea and export pod and owner metrics. So the energy metrics can be aggregated at the query time. Would that work?

marceloamaral, 2022-10-19:
> having a node watcher on each node requires additional resources and privileges.

Right!
I think the question is: how much resource?
A part of the API server access permission, what other privilege?

Having a watcher on each node will give some other advantages, like reducing the kubelet overhead and ensuring consistency in our data by always removing the pods that were deleted.

I agree that scalability is something we need to understand very well!
Let's try to investigate the possible impact of watchers on each node....

> I'm thinking of creating another exporter

This could be a solution.
But I'm afraid we will create too many external dependencies for Kepler and in the end it might be hard to understand the flow.... And we loose the consistency advantage that I mentioned before...

marceloamaral, 2022-10-19:
I'll ask some K8s API server scalability experts about this!

rootfs, 2022-10-21:
The dependence issue is double edged, the pod owner exporter can be used for other purposes as well (e.g. for metrics-server). By separating it from kepler, we have the flexibility to deploy it only when the aggregation is needed.

simonpasquier, 2022-11-25:
I don't have a lot of context but have you considered joining the kepler metrics with kube-state-metrics (KSM) metrics. In particular, KSM already provides the pod -> replicaset -> deployment relationships. And you even have recording rules in https://github.com/kubernetes-monitoring/kubernetes-mixin to identify which pods belong to which deployment/daemonset/statefulset.

rootfs, 2022-11-27:
@simonpasquier thanks for the pointer, I glanced KSM, it looks the metrics are based k8s objects, not object metrics that are already in prometheus. Would you share more info about how to aggregating e.g. deployment level metrics?

closed 2023-05-24 by stale[bot]