sustainable-computing-io/kepler #304: 5s is not enough for e2e test on Unit test Mac

sunya-ch, 2022-10-17:
**Describe the bug**
A clear and concise description of what the bug is.

The Unit test Mac keeps failing on e2e test (while the other unit tests were fine) when fixing first waiting time to 5s.
https://github.com/sustainable-computing-io/kepler/actions/runs/3262694289/jobs/5360198723

**To Reproduce**
Steps to reproduce the behavior:
Not trivial but it happens only on github actions when I add some initial logics of power model before starting the Kepler exporter server.

**Expected behavior**
A clear and concise description of what you expected to happen.
The test should wait until the server is ready or the program is failed and terminated.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Linux (but test pass in local)
 - Browser: Safari

**Additional context**
I added kFlag as a keyword for Eventually function to periodically (every 5s for 1 minute) check the status instead of fixing waiting time and now, it can pass the test at 25s.
Note that, in the other e2e unit test (except Mac), the kepler is ready right after the first check (at 5s).

```golang
Eventually(keplerSession.Out, 30, 5).Should(gbytes.Say(kFlag))
```
kFlag commit: https://github.com/sustainable-computing-io/kepler/pull/225/commits/15b693b5a246955b6103522a35dcecb5f3de0e12
pass test result: https://github.com/sustainable-computing-io/kepler/actions/runs/3266241292/jobs/5369687391

p.s. I can separate kFlag commit for another PR if needed.

marceloamaral, 2022-10-18:
As we discussed before, we don't need the string to say kepler failed, but to say it has started...
And then the test will try the health check to see if the kepler is still alive...
So, what we want then is a log that prints after the server is started

We can fix that in another PR....

sunya-ch, 2022-10-18:
If we use eventually without reporting when the kepler failed and exits, it can wait forever.

marceloamaral, 2022-10-18:
This should be fine, 30s is a very short timeout interval
Making the logic more transparent is better for this case

marceloamaral, 2022-10-18:
But we can check both, if it failed or succeeded

marceloamaral, 2022-10-18:
I created the PR #307 to fix this

marceloamaral, 2022-10-25:
In fact, PR #307 fixed this...

The health check should only start responding when kepler is fully initialized

closed 2022-10-25 by marceloamaral