sustainable-computing-io/kepler #310: Improve GPU power estimate using perf counters

rootfs, 2022-10-18:
**Is your feature request related to a problem? Please describe.**
Current GPU power estimate is based on memory footprint. This model doesn't account for factors such as GPU instructions, temperature, or frequency. 

**Describe the solution you'd like**
As studied in [1] with (old generation of) NVIDIA GPU and [2] with AMD GPU, selective performance counters are highly reliable indicators of power usage. We should review these studies and improve GPU power estimate.

1. S. Song, C. Su, B. Rountree and K. W. Cameron, "A Simplified and Accurate Model of Power-Performance Efficiency on Emergent GPU Architectures," 2013 IEEE 27th International Symposium on Parallel and Distributed Processing, 2013, pp. 673-686, doi: 10.1109/IPDPS.2013.73.
2. G. Wu, J. L. Greathouse, A. Lyashevsky, N. Jayasena and D. Chiou, "GPGPU performance and power estimation using machine learning," 2015 IEEE 21st International Symposium on High Performance Computer Architecture (HPCA), 2015, pp. 564-576, doi: 10.1109/HPCA.2015.7056063.

rootfs, 2022-10-18:
@sustainable-computing-io/maintainer

marceloamaral, 2022-10-19:
The nvidia library shows the memory usage and gpu utilization regarding the compute part....
I am planning to update that, will submit a PR soon!

rootfs, 2022-10-19:
@marceloamaral I think current model works for time-sharing case (e.g. on GKE), in the memory-sharing (e.g. AliCloud) case, more robust model is needed.

marceloamaral, 2023-05-18:
Since we are using the GPU utilization metric I am closing this issue.
We can reopen if needed.

closed 2023-05-18 by marceloamaral