sustainable-computing-io/kepler #312: Ephemeral Pod is not

nikimanoledaki, 2022-10-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
I used ARC to create a self-hosted runner for GitHub Actions. It runs for 10-15s and then self-destructs. Follow [these steps](https://github.com/nikimanoledaki/sustainability-journey-with-gitops/actions/workflows/self-destruct.yaml) to re-create this.

Querying Kepler for this Pod does not return any data. Kepler sometimes sees the Pod and sometimes does not (@rootfs can provide more info).

**Expected behavior**
Expected Pod to be consistently picked up by Kepler

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] RHEL 8

marceloamaral, 2022-10-20:
Since you are using GitHub Actions, the environment is probably a VM....
But, can you check the logs to see if the BPF module is running?

jichenjc, 2022-10-20:
I reported https://github.com/sustainable-computing-io/kepler/issues/264 before
not sure we fixed all those or still some occurance might lead to this issue?

nikimanoledaki, 2022-10-20:
Thanks both, I am using self hosted runners with ARC, where the runners run as Pods in the cluster so they are not VMs in this case. I included a link for how I set them up in case you would like to try the same setup :) When the action is triggered, the tests run for approx 10-15s, so they go very quickly and perhaps that is why Kepler does not detect and register them.

nikimanoledaki, 2022-10-20:
Sorry, just realised I did not add a link to the setup, here it is: https://gist.github.com/nikimanoledaki/545b215c229ebe4ed6ad1d8c189ecbb9

marceloamaral, 2022-10-20:
>the tests run for approx 10-15s

But how long does the pod stay alive?
Kepler collects metrics every 3s, so in theory if pods stay at 10s we should see them

nikimanoledaki, 2022-10-20:
Given that each self-destructing Pod is created as soon as the previous self-destructing Pod terminates, and given that I start the test as soon the previous test ends, each Pod should run only for the duration of the test, so 10-15s.

rootfs, 2022-10-21:
Just a thought, would upgrading bcc library fix this issue? #317

marceloamaral, 2023-05-18:
We have done several improvements. Is this still an issue?

nikimanoledaki, 2023-07-30:
Closing this since I do not have access to the environment/project that led to this issue.. If a similar issue occurs in the future, we can refer to this one for more info.

closed 2023-07-30 by nikimanoledaki