sustainable-computing-io/kepler #317: time to upgrade to latest bcc-devel pkg?

rootfs, 2022-10-21:
**Is your feature request related to a problem? Please describe.**
Current bcc-devel pkg used by Kepler is [v0.19](https://github.com/sustainable-computing-io/kepler/blob/main/build/Dockerfile.base#L7).
 
There are [quite changes](https://github.com/iovisor/bcc/compare/v0.24.0..v0.19.0?diff=unified) in latest iovisor bcc since v0.19. 

Is it time to upgrade to [latest](http://mirror.centos.org/centos/8-stream/PowerTools/x86_64/os/Packages/bcc-devel-0.24.0-2.el8.x86_64.rpm) bcc-devel pkg

**Describe the solution you'd like**
Upgrade the kepler base image

jichenjc, 2022-10-21:
maybe you cut a release like 0.1 or 0.01 of kepler with current bcc then 
continue to update main with latest bcc so folks won't be affected if they need older one somehow ?

marceloamaral, 2022-10-21:
Is there any kernel requirement to upgrade the bcc?
If not, this should be fine!

rootfs, 2022-10-21:
@jichenjc there is a `release-0.3` branch. 
@marceloamaral newer bcc supports newer kernels per release page

rootfs, 2022-10-21:
Here is the bcc release info https://github.com/iovisor/bcc/releases

marceloamaral, 2022-10-21:
@rootfs with a quick look to the releases minutes I did not see any possible conflict...
We can try, I should be fine

PR #321 (2022-10-24): kepler base, build, and release images: upgrade to ubi 8.6, bcc 0.24.0, nvidia nvml 520
fix #317
closed 2022-10-24 by marceloamaral