sustainable-computing-io/kepler #318: Cannot deploy Kepler with make cluster-sync

marceloamaral, 2022-10-21:
**Describe the bug**
After merging PR #311 the kepler deployment is not working anymore

```
kustomize manifests...
+ kubectl kustomize _output/manifests/kubernetes/generated/bm
Error: rawResources failed to read Resources: Load from path ../base failed: '../base' must be a file (got d='/home/mamaral/go/src/github.com/marceloamaral/kepler/_output/manifests/kubernetes/generated/base')
```

**To Reproduce**
Steps to reproduce the behavior:
` make cluster-sync`

jichenjc, 2022-10-21:
yes, I found this , working on this now ,sorry for inconvenience

marceloamaral, 2022-10-21:
Thanks @jichenjc for the quick reply...

Don't worry, but I am wondering why the tests did not get this....

jichenjc, 2022-10-21:
let me fix the issue first then find why test not find it

marceloamaral, 2022-10-21:
@jichenjc the problem is the `kubectl` version: [here](https://stackoverflow.com/questions/67071962/kubectl-apply-k-throws-error-rawresources-failed-to-read-resources-load-from ). That's why we didn't see this in the CI...

After updating my `kubectl`, the deployment process is working now...

Unfortunately I don't know which version I had before... but we should document this dependency somewhere

jichenjc, 2022-10-21:
Thanks for sharing this, I found something still need improve, will submit a PR soon

marceloamaral, 2022-10-21:
But the `make cluster-sync` is still not working....

For the CI, I upload the new built image to the kind cluster...
In order to use this image, we need to change the manifest to use the new image....[here](https://github.com/sustainable-computing-io/kepler/blob/main/hack/build-manifest.sh#L43)

But the image name is not being updated now...

jichenjc, 2022-10-21:
yes, I found that bug, working on that now .. will submit a PR soon

jichenjc, 2022-10-21:
looks like the deployment of make cluster-sync is different to manifest.. need update them as well

marceloamaral, 2022-10-21:
Here is how to fix it:
```bash
if [[ $CLUSTER_PROVIDER == "openshift" ]]; then
    cat manifests/${CLUSTER_PROVIDER}/kepler/01-kepler-install.yaml | sed "s|image:.*|image: $IMAGE_REPO:$IMAGE_TAG|" > ${MANIFESTS_OUT_DIR}/kepler/01-kepler-install.yaml
else
    cat ${MANIFESTS_OUT_DIR}/bm/deployment.yaml | sed "s|image:.*|image: $IMAGE_REPO:$IMAGE_TAG|" > tmp_file
    mv tmp_file ${MANIFESTS_OUT_DIR}/bm/deployment.yaml
    cat ${MANIFESTS_OUT_DIR}/vm/deployment.yaml | sed "s|image:.*|image: $IMAGE_REPO:$IMAGE_TAG|" > tmp_file 
    mv tmp_file ${MANIFESTS_OUT_DIR}/vm/deployment.yaml
fi
```

I updated for Kubernetes, but you need to also update it for openshift

marceloamaral, 2022-10-21:
You just need to update the `hack/build-manifest.sh`

jichenjc, 2022-10-21:
https://github.com/sustainable-computing-io/kepler/pull/319/ is proposed for this , thanks for report and hope that works

PR #319 (2022-10-21): fix can't make cluster-up/sync
I used following to fix #318 

```
CLUSTER_PROVIDER="kind" make cluster-up
CLUSTER_PROVIDER="kind" make cluster-sync

I can see this 

# kubectl get pods -A
NAMESPACE            NAME                                         READY   STATUS    RESTARTS   AGE
kepler               kepler-exporter-g7qd8                        1/1     Running   0          2m55s
kube-system          coredns-565d847f94-fjg2t                     1/1     Running   0          5m12s
kube-system          coredns-565d847f94-llwwh                     1/1     Running   0          5m12s
kube-syste [... cut]
closed 2022-10-21 by rootfs