sustainable-computing-io/kepler #320: Refactor and rearrange the packages and some names

marceloamaral, 2022-10-21:
**Is your feature request related to a problem? Please describe.**

Kepler code is growing and it is starting to be difficult to follow the structures.
So we should refactor and rearrange the packages and some names.

What should we do:
- [ ] move all bpf related files to just one place
- [ ] move all cgroup related files to just one place
- [ ] create a general cache structure that the collector and model packages will use to write/read data, eliminating unnecessary collections
- [ ] refactor the reader to more meaningful filenames and distribute it to different files
- [ ] create a new package to manage/control the system
- [ ] do not mix cgroup with kubelet files. These are different ways to collect metrics/information
- [ ] rename gpu package to more general resource name like accelerators
- [ ] move model related data to fixtures folder (commonly used in golang projects)
- [ ] some projects use the name exporter instead of collector.....not sure which is more intuitive...

------
**Describe the solution you'd like**
The following structure was inspired in the [cadvisor](https://github.com/google/cadvisor), [node_exporter](https://github.com/prometheus/node_exporter), [openstack-exporter](https://github.com/openstack-exporter/openstack-exporter) among others....
```bash
.
├── bpf **(move all bpf related files to here)**
│   ├── assets.go
│   └── perf_event_bindata.go
│   └── perf_event.c
│   ├── attacher
│   │   ├── bcc_attacher.go
│   │   ├── bcc_attacher_stub.go
│   │   ├── bpf_perf.go
├── cgroup **(move all cgroup related files to here)**
│   ├── resolve_container.go
│   ├── cgroup_stats.go
│   ├── cgroup_stats_test.go
│   ├── path_handler.go
│   ├── path_handler_test.go
│   ├── slice_handler.go
│   ├── slice_handler_test.go
│   ├── stat_reader.go
│   ├── stat_reader_test.go
│   ├── suite_test.go
│   └── test
├── cache **(general structure that collector and model will use to write/read data)**
│   ├── cache.go
├── metric  **(move most of the collector metric stat definition to here)**
│   ├── container_metric.go
│   ├── node_metric.go
│   ├── stats.go
│   ├── utils.go
│   └── utils_test.go
├── collector **(refactor the reader to more meaningful file names)**
│   ├── collector.go
│   ├── collector_container_bpf.go
│   ├── collector_container_cgroup.go
│   ├── collector_container_energy.go
│   ├── collector_node_energy.go
│   ├── suite_test.go
│   └── types.go
├── config
│   ├── config.go
│   ├── config_test.go
│   ├── suite_test.go
│   └── types.go
├── manager **(new package to manager/control the system)**
│   ├── manager.go
├── model
│   ├── fixtures **(move model related data to here)**
│   │   │   ├── power_data.csv
│   │   │   ├── power_model.csv
│   │   │   ├── etc...
│   ├── estimator
│   │   ├── local
│   │   │   ├── lr.go
│   │   │   ├── lr_test.go
│   │   │   ├── ratio.go
│   │   │   └── ratio_test.go
│   │   └── sidecar
│   │       ├── estimate.go
│   │       └── estimate_test.go
│   ├── model.go
│   ├── model_test.go
│   ├── node_component_power.go
│   ├── node_total_power.go
│   ├── pod_power.go
│   ├── suite_test.go
│   └── types
│       └── types.go
├── kubelet **(keep only kubelet related files here)**
│   ├── byte_order.go
│   ├── byte_order_test.go
│   ├── kubelet_pod_lister.go
│   ├── pod_lister_test.go
│   ├── suite_test.go
│   ├── util_darwin.go
│   └── util_linux.go
├── power
│   ├── acpi
│   │   └── acpi.go
│   ├── accelerators **(rename to more general resource)**
│   │   ├── gpu_dummy.go
│   │   └── gpu_nvml.go
│   └── rapl
│       ├── power.go
│       └── source
│           ├── dummy.go
│           ├── estimate.go
│           ├── msr.go
│           ├── msr_util.go
│           ├── package_energy.go
│           ├── sysfs.go
│           └── sysfs_util.go
└── utils
    └── utils.go
```

**Additional context**
The current structure is:
```bash
├── attacher
│   ├── bcc_attacher.go
│   ├── bcc_attacher_stub.go
│   └── bpf_perf.go
├── bpfassets
│   ├── assets.go
│   └── perf_event_bindata.go
├── cgroup
│   ├── path_handler.go
│   ├── path_handler_test.go
│   ├── slice_handler.go
│   ├── slice_handler_test.go
│   ├── stat_reader.go
│   ├── stat_reader_test.go
│   ├── suite_test.go
├── collector
│   ├── collector.go
│   ├── collector_test.go
│   ├── metrics.go
│   ├── metrics_test.go
│   ├── node_energy.go
│   ├── pod_energy.go
│   ├── reader.go
│   ├── suite_test.go
│   └── types.go
├── config
│   ├── config.go
│   ├── config_test.go
│   ├── suite_test.go
│   └── types.go
├── model
│   ├── estimate.go
│   ├── estimate_test.go
│   ├── estimator
│   │   ├── local
│   │   │   ├── lr.go
│   │   │   ├── lr_test.go
│   │   │   ├── ratio.go
│   │   │   └── ratio_test.go
│   │   └── sidecar
│   │       ├── estimate.go
│   │       └── estimate_test.go
│   ├── model.go
│   ├── model_test.go
│   ├── node_component_power.go
│   ├── node_total_power.go
│   ├── pod_power.go
│   ├── suite_test.go
│   └── types
│       └── types.go
├── podlister
│   ├── byte_order.go
│   ├── byte_order_test.go
│   ├── cgroup_stats.go
│   ├── cgroup_stats_test.go
│   ├── kubelet_pod_lister.go
│   ├── pod_lister_test.go
│   ├── resolve_container.go
│   ├── suite_test.go
│   ├── util_darwin.go
│   └── util_linux.go
├── power
│   ├── acpi
│   │   └── acpi.go
│   ├── gpu
│   │   ├── gpu_dummy.go
│   │   └── gpu_nvml.go
│   └── rapl
│       ├── power.go
│       └── source
│           ├── dummy.go
│           ├── estimate.go
│           ├── msr.go
│           ├── msr_util.go
│           ├── package_energy.go
│           ├── sysfs.go
│           └── sysfs_util.go
└── utils
    └── utils.go
```
----
Btw, the PR #287 will address some of this proposal:
- [x] refactor the reader to more meaningful filenames and distribute it to different files
- [x] create a new package to manage/control the system

marceloamaral, 2022-10-24:
/cc @rootfs @sunya-ch

sunya-ch, 2022-10-26:
Good job. 
In my opinion, there might be too many folder on level-1 and can be a little bit difficult to navigate the core code. How's about put  metric sources (cgroup, kubelet, BPF,...) together under a folder.

marceloamaral, 2023-05-18:
We have done almost everything here. So I am closing

closed 2023-05-18 by marceloamaral