sustainable-computing-io/kepler #329: aggregate energy consumption from Pod level to higher level

rootfs, 2022-10-24:


rootfs, 2023-01-24:
@marceloamaral join metrics at the Prometheus level. Investigate the relationship in the metadata. 
@sunya-ch not to query API server from every nodes. A controller (along with Kepler in the deployment) to expose high level API and Pods relationship to Prometheus.  Aggregation is done at Prometheus. 
@wangchen615 metrics needs to be indexed for Prometheus query. Namespace, deployment, pods....

rootfs, 2023-01-24:
1) what metadata are already in Prometheus (e.g. Job)

marceloamaral, 2023-05-18:
`kube-state-metric` exports the owner of objects.
So the pod created by a Job or Deployment will have this information exported.

@sunya-ch did quick test to aggregate the metrics a it works:
[image]

marceloamaral, 2023-09-04:
I have created a side project to help with that, which export labels to be used to aggregate the metrics of different project, executions, etc.
https://github.com/marceloamaral/label-exporter

The plan is to migrate the label-exporter inside the sustainable-computing-io and also expose object ownership metrics as kube-state-metric.

rootfs, 2023-09-20:
this can be resolved with label exporter provided by @marceloamaral

closed 2023-09-20 by rootfs