sustainable-computing-io/kepler #331: support more kubernetes/openshift cluster types

rootfs, 2022-10-24:


SamYuan1990, 2022-10-25:
microshift？

sallyom, 2022-11-01:
PR for CI with OpenShift: https://github.com/openshift/release/pull/32639

sallyom, 2022-11-01:
@rootfs please assign me to this card @sallyom thank you

sallyom, 2022-11-01:
> how to track microshift and make it available on the CI

I believe we can utilize the openshift/release MicroShift CI, then add Kepler deployment to that - (and also add documentation to Kepler repo wrt how a user can deploy MicroShift on their own)

sallyom, 2022-11-07:
I've been putting together the recipe for running kepler/prom/grafana in MicroShift, with KVM, will open a PR to document this

marceloamaral, 2022-11-08:
Does the MicroShift need to be run in KVM? It would be better to run it directly in Bare-Metal.

In the case you will run in KVM, it could also be possible to enable RAPL and Hardware Counter metrics.
This is related to the issue #367.

closed 2023-05-24 by stale[bot]